OpenCPN/OpenCPN #4167: OpenCPN 5.10.2 doesn't start when using i3 as the window manager

invader-zimm, 2024-10-12:
**Describe the bug**
I'm using Artix Linux and the i3 window manager, and the latest version of OpenCPN installed through flatpak doesn't start. Well, according to the log it starts but exits immediately without any errors. This is what the log says:
```
13:19:42.254 MESSAGE ocpn_app.cpp:1948 opencpn::MyApp starting exit.
13:19:42.256 MESSAGE ocpn_app.cpp:1988 LOGBOOK:  2024-10-12 17:19:42 UTC OFF: Lat    0.00000 Lon    0.00000
13:19:42.256 MESSAGE ocpn_app.cpp:2016 opencpn::MyApp exiting cleanly...
```

It looks like a regression from 5.8.4. I downgraded OpenCPN by running `flatpak update --commit=c7130829af3b2dc02a6d2f17fd266e62c0003dd3a33d19fd5565b3b664c542f4 org.opencpn.OpenCPN` and version 5.8.4 works without issues.

However, on the same machine I also have MATE, and when using that as the window manager OpenCPN 5.10.2 starts just fine... The problem occurs only with i3wm. I was thinking maybe something is not configured properly on my system but the previous release works so I'm not sure what might be misconfigured.

Any idea how to debug this? The log file doesn't really say anything and there are no errors printed out anywhere.

**Expected behavior**
OpenCPN starts and runs like in previous releases.

invader-zimm, 2024-10-26:
I have a bit more info, I compiled OpenCPN from AUR and everything works as expected with i3wm, it starts without issues. Comparing the log file between the local build (which starts just fine) and the flatpak one I see that one of the first thing the local build does is look for wxWidgets (i have version 3.2.6 Linux 64 bit wxGTK). The flatpak build is completely missing that line from the logs, so my assumption is that for some reason it doesn't detect wxWidgets and then it exits.

leamas, 2024-10-27:
@invader-zimm : 

hm... Could you first please share the first 10 lines or so from  *~/.var/app/org.opencpn.OpenCPN/config/opencpn/opencpn.log*?

Then, try to list   the wxWidgets libraries installed using `flatpak run --command=ls org.opencpn.OpenCPN//stable /app/lib | grep wx`  and share the results

invader-zimm, 2024-10-27:
The log from _~/.var/app/org.opencpn.OpenCPN/config/opencpn/_ contains only this, nothing else:
```
12:38:03.244 MESSAGE ocpn_app.cpp:1948 opencpn::MyApp starting exit.
12:38:03.244 MESSAGE ocpn_app.cpp:1988 LOGBOOK:  2024-10-27 16:38:03 UTC OFF: Lat    0.00000 Lon    0.00000
12:38:03.244 MESSAGE ocpn_app.cpp:2016 opencpn::MyApp exiting cleanly...
```

The log from _~/.opencpn_ which is generated by the AUR build (the one that starts without issues) says:
```
12:27:22.496 MESSAGE ocpn_app.cpp:1126 

________

12:27:22.496    INFO logger.cpp:130 ------- OpenCPN version 5.10.2 restarted at 2024-10-27 -------

12:27:22.496    INFO logger.cpp:130 Using loglevel INFO
12:27:22.496 MESSAGE ocpn_app.cpp:1149 wxWidgets version: wxWidgets 3.2.6 Linux 64 bit wxGTK
12:27:22.502 MESSAGE ocpn_app.cpp:1154 OS Version reports as:  6.6
12:27:22.502 MESSAGE ocpn_app.cpp:1156 MemoryStatus:  mem_total: 7225 mb,  mem_initial: 1012 mb
12:27:22.502 MESSAGE ocpn_app.cpp:1168 OCPN_OSDetail:  x86_64 ; Artix Linux ; rolling ; artix ; 
12:27:22.502 MESSAGE ocpn_app.cpp:1172 SData_Locn is /usr/share/opencpn/
12:27:22.502 MESSAGE ocpn_app.cpp:1190 PrivateDataDir is /home/user/.opencpn
12:27:22.503 MESSAGE ocpn_app.cpp:1255 Using existing Config_File: /home/user/.opencpn/opencpn.conf
12:27:22.507 MESSAGE navutil.cpp:1247 winPluginDir, read from ini file: 
12:27:22.508 MESSAGE navutil.cpp:1329 Setting Ownship Lat/Lon 45.5521, -73.5716
12:27:22.508 MESSAGE navutil.cpp:3327 Switch InlandEcdis mode Off
12:27:22.509 MESSAGE ConfigMgr.cpp:548 Loading Configs catalog: /home/user/.opencpn/Configs/configs.xml
12:27:22.513 MESSAGE styles.cpp:841 Styles loading from /usr/share/opencpn/uidata/styles.xml
12:27:22.515 MESSAGE styles.cpp:818 No styles found at: /home/user/
12:27:22.515 MESSAGE styles.cpp:818 No styles found at: /home/user/.opencpn/
12:27:22.528 MESSAGE styles.cpp:775 Using SVG Icons
12:27:22.528 MESSAGE ocpn_app.cpp:1316 Detected display size (horizontal): 309 mm
12:27:22.528 MESSAGE OCPNPlatform.cpp:1022 Getting DefaultSystemLocale...
12:27:22.530 MESSAGE ocpn_app.cpp:1354 System default Language:  en_US
...
```

Listing the wxWidgets libs from the flatpak build returns this:
libwx_baseu-3.2.so
libwx_baseu-3.2.so.0
libwx_baseu-3.2.so.0.2.2
libwx_baseu_net-3.2.so
libwx_baseu_net-3.2.so.0
libwx_baseu_net-3.2.so.0.2.2
libwx_baseu_xml-3.2.so
libwx_baseu_xml-3.2.so.0
libwx_baseu_xml-3.2.so.0.2.2
libwx_gtk3u_adv-3.2.so
libwx_gtk3u_adv-3.2.so.0
libwx_gtk3u_adv-3.2.so.0.2.2
libwx_gtk3u_aui-3.2.so
libwx_gtk3u_aui-3.2.so.0
libwx_gtk3u_aui-3.2.so.0.2.2
libwx_gtk3u_core-3.2.so
libwx_gtk3u_core-3.2.so.0
libwx_gtk3u_core-3.2.so.0.2.2
libwx_gtk3u_gl-3.2.so
libwx_gtk3u_gl-3.2.so.0
libwx_gtk3u_gl-3.2.so.0.2.2
libwx_gtk3u_html-3.2.so
libwx_gtk3u_html-3.2.so.0
libwx_gtk3u_html-3.2.so.0.2.2
libwx_gtk3u_media-3.2.so
libwx_gtk3u_media-3.2.so.0
libwx_gtk3u_media-3.2.so.0.2.2
libwx_gtk3u_propgrid-3.2.so
libwx_gtk3u_propgrid-3.2.so.0
libwx_gtk3u_propgrid-3.2.so.0.2.2
libwx_gtk3u_qa-3.2.so
libwx_gtk3u_qa-3.2.so.0
libwx_gtk3u_qa-3.2.so.0.2.2
libwx_gtk3u_ribbon-3.2.so
libwx_gtk3u_ribbon-3.2.so.0
libwx_gtk3u_ribbon-3.2.so.0.2.2
libwx_gtk3u_richtext-3.2.so
libwx_gtk3u_richtext-3.2.so.0
libwx_gtk3u_richtext-3.2.so.0.2.2
libwx_gtk3u_stc-3.2.so
libwx_gtk3u_stc-3.2.so.0
libwx_gtk3u_stc-3.2.so.0.2.2
libwx_gtk3u_xrc-3.2.so
libwx_gtk3u_xrc-3.2.so.0
libwx_gtk3u_xrc-3.2.so.0.2.2
wx
```

Not sure, if wxWidgets is not the issue maybe there's a problem with OpenGL initialization?

invader-zimm, 2024-10-27:
Running `flatpak run --command=bash org.opencpn.OpenCPN//stable` doesn't do anything, it doesn't print anything in the console or in the log file.

leamas, 2024-10-27:
Your last comment seems to shed some light on this.

> Authorization required, but no authorization protocol specified

No idea what this could be about.  A quick google indicates that this might be about *~/.Xauthority*, but without a test case I'm lost. 

> 12:57:59: Error: Unable to initialize GTK+, is DISPLAY set properly?

Well, well... what happens if you run `opencpn-glutil opengl-info`  outside  of and within the sandbox? Please only share the last ten lines of output from each run.

invader-zimm, 2024-10-27:
From within the sandbox I get
```
$> flatpak run --command=bash org.opencpn.OpenCPN//stable
$> opencpn-glutil opengl-info
Authorization required, but no authorization protocol specified

14:11:18: Error: Unable to initialize GTK+, is DISPLAY set properly?
```

From outside the sandbox (AUR build) I get
```
$> opencpn-glutil opengl-info
   "GL_EXT_framebuffer_object" : 1,
   "GL_RENDERER" : "Mesa Intel(R) HD Graphics 520 (SKL GT2)",
   "GL_ARB_texture_non_power_of_two" : 1,
   "GL_SHADING_LANGUAGE_VERSION" : "4.60",
   "GL_ARB_texture_rectangle" : 1,
   "GL_EXTENSIONS" : [
      "GL_ARB_multisample",
      "GL_EXT_abgr",
      "GL_EXT_bgra",
      "GL_EXT_blend_color",
      "GL_EXT_blend_minmax",
      "GL_EXT_blend_subtract",
      "GL_EXT_copy_texture",
      "GL_EXT_subtexture",
      "GL_EXT_texture_object",
      "GL_EXT_vertex_array",
      "GL_EXT_compiled_vertex_array",
      "GL_EXT_texture",
      "GL_EXT_texture3D",
      "GL_IBM_rasterpos_clip",
      "GL_ARB_point_parameters",
      "GL_EXT_draw_range_elements",
      "GL_EXT_packed_pixels",
      "GL_EXT_point_parameters",
      "GL_EXT_rescale_normal",
      "GL_EXT_separate_specular_color",
      "GL_EXT_texture_edge_clamp",
      "GL_SGIS_generate_mipmap",
      "GL_SGIS_texture_border_clamp",
      "GL_SGIS_texture_edge_clamp",
      "GL_SGIS_texture_lod",
      "GL_ARB_framebuffer_sRGB",
      "GL_ARB_multitexture",
      "GL_EXT_framebuffer_sRGB",
      "GL_IBM_multimode_draw_arrays",
      "GL_IBM_texture_mirrored_repeat",
      "GL_3DFX_texture_compression_FXT1",
      "GL_ARB_texture_cube_map",
      "GL_ARB_texture_env_add",
      "GL_ARB_transpose_matrix",
      "GL_EXT_blend_func_separate",
      "GL_EXT_fog_coord",
      "GL_EXT_multi_draw_arrays",
      "GL_EXT_secondary_color",
      "GL_EXT_texture_env_add",
      "GL_EXT_texture_filter_anisotropic",
      "GL_EXT_texture_lod_bias",
      "GL_INGR_blend_func_separate",
      "GL_NV_blend_square",
      "GL_NV_light_max_exponent",
      "GL_NV_texgen_reflection",
      "GL_NV_texture_env_combine4",
      "GL_S3_s3tc",
      "GL_SUN_multi_draw_arrays",
      "GL_ARB_texture_border_clamp",
      "GL_ARB_texture_compression",
      "GL_EXT_framebuffer_object",
      "GL_EXT_texture_compression_s3tc",
[... 243 lines cut ...]
      "GL_EXT_semaphore_fd",
      "GL_KHR_parallel_shader_compile",
      "GL_NV_alpha_to_coverage_dither_control",
      "GL_EXT_EGL_image_storage",
      "GL_EXT_shader_framebuffer_fetch_non_coherent",
      "GL_EXT_texture_shadow_lod",
      "GL_INTEL_blackhole_render",
      "GL_INTEL_shader_atomic_float_minmax",
      "GL_INTEL_shader_integer_functions2",
      "GL_MESA_framebuffer_flip_y",
      "GL_NV_compute_shader_derivatives",
      "GL_EXT_EGL_sync",
      "GL_EXT_demote_to_helper_invocation",
      "GL_EXT_EGL_image_storage_compression",
      "GL_NV_ES1_1_compatibility",
      "GL_MESA_texture_const_bandwidth"
   ],
   "GL_ERROR" : 0,
   "GL_USABLE" : true,
   "GL_VENDOR" : "Intel",
   "GL_ERROR_STRING" : "No error",
   "GL_VERSION" : "4.6 (Compatibility Profile) Mesa 24.2.5-arch1.1",
   "GL_OES_texture_npot" : 0
}
```

leamas, 2024-10-27:
OK, this somehow nails the problem. 

I'm on very thin ice here. That said, the first things is probably to get rid of the "Authorization required" message.  Google shows a lot of matches, but without a test case I cannot evaluate much. Can you?

leamas, 2024-10-27:
BTW: When starting up, opencpn runs  opencpn-glutil in order to establish that the opengl environment is OK which it obviously isn't here. This is probably the root cause here.

Have you tried to run without opengl? `flatpak run org.opencpn.OpenCPN//stable --no_opengl`

nohal, 2024-10-27:
If `opencpn-glutil` fails, the main application starts just fine and disables OpenGL, that's it's purpose. The problem here does not seem to have anything to do with OpenGL, it seems that in flatpak runtime, the window manager is not accessible at all.
Probably easy to check if it a systemwide issue by trying to run some other GUI application flatpak.

invader-zimm, 2024-10-27:
Running without OpenGL does the same thing, the program doesn't start.

I'll try to fix the authorization issue, BUT: I ran `opencpn-glutil opengl-info` in the sandbox under MATE Desktop, on which flatpak OpenCPN does start, and I got exactly the same errors regarding authorization and DISPLAY. I'm not sure it will help but I'll try to get rid of those errors, and see about other apps as well.

leamas, 2024-10-27:
> Running without OpenGL does the same thing, the program doesn't start.

this possibly points in another direction. If you start with `flatpak run org.opencpn.OpenCPN//stable --no_opengl --loglevel debug`, does the log then say something more meaningful?

leamas, 2025-04-27:
Ping? 

Without more info we will need to close this bug

invader-zimm, 2025-05-16:
Sorry I haven't made any progress on this. No matter what I tried it didn't work. I ended up compiling from source, and that worked so I'll keep doing that.
If no one else encountered this issue, it can be closed.

closed 2025-05-16 by invader-zimm