OpenCPN/OpenCPN #4168: 5.10.2 could not zoom below 100m (50m and 20m not available)

yann78, 2024-10-15:
When I set the anchor watch I set the zoom to 20m to get an accurate view.

These levels are no more available on the two 5.10.x versions

see below 5.8.4-2
[image]

 Pi4 
  Operating System: Debian GNU/Linux 11 (bullseye)
   Kernel: Linux 6.1.32-v8+
   Architecture: arm64

Hakansv, 2024-10-15:
yann78..
This is as designed so the program is working as it should. So obviously not a bug.

For more info how to control it if you'd like to change, see this [CF post](https://www.cruisersforum.com/forums/f134/overzoom-wanted-237058.html) That user is on Windows so for your system:
Search for MaxZoomScale in opencpn.conf and change to your desired setting.

yann78, 2024-10-15:
Many thanks, it works.
I was surprised by  this new standard setting

closed 2024-10-15 by yann78

Hakansv, 2024-10-16:
Good it works for you. But what I know this is not so new. The limitation has been in the code "for ever". The config parameter was added in 2021.