OpenCPN/OpenCPN #4170: macOS 15.x (Sequoia) - higher hurdles to install OpenCPN

bcn2, 2024-10-18:
With 15.x it is not any longer possible to install OpenCPN from the finder right-clicking the .pkg and confirming to "install anyhow"
One gets urged to erase the downloaded "suspicious code" and has to pass through the macOS Preferences for Privacy&Security, there "Security" to find an entry of the "blocked download" and release from there.
The steps to take are hidden under the question mark ("?" or help) of the first pop-up.

Perhaps too complicated for the unprepared user.

the screen shots are from a spanish version, but what happens from now on should be clear.
[image]

 
[image]

[image]

bcn2, 2024-10-18:
Yes, this the question - worthwhile to make the life easier for the user?

antipole2, 2024-10-28:
The simplest is for the individual user to override Gatekeeper.  But this is inconvenient and some users may find it difficult.

> Method 1: Open via System Settings (One-Time Bypass)
> If you attempt to open an unsigned app and it’s blocked, macOS may still give you a way to bypass this restriction in System Settings:
> 
> Try to open the app as you usually would by double-clicking it. A pop-up will likely inform you that the app can’t be opened because it’s from an unidentified developer.
> Go to System Settings > Privacy & Security.
> Scroll down until you see a message about the app that was blocked from opening.
> Click Open Anyway next to the app name.
> After doing this, you should be able to open the app normally, and this permission is usually retained.
> Method 2: Use Terminal to Remove Quarantine Attribute
> You can manually remove the “quarantine” attribute that macOS applies to downloaded apps:
> 
> Open Terminal.
> Use the following command to remove the quarantine attribute, replacing the path to the app file as needed:
> bash
> sudo xattr -rd com.apple.quarantine /path/to/app.app
> Enter your password when prompted.
> After this, you should be able to open the app as usual.
> Method 3: Enable Boot into Reduced Security Mode (Not Recommended for Everyday Use)
> This is a more advanced approach that can affect overall system security but allows the installation of any unsigned app. It requires booting into macOS Recovery:
> 
> Restart your Mac and hold down Command + R until it boots into Recovery Mode.
> Go to the Utilities menu and open Terminal.
> Type in the command:
> bash
> spctl kext-consent add <TeamID>
> Reboot the system and try installing the app again.

Better would be for a developer to sign the binary before release.

> 1. Enroll in the Apple Developer Program
> To obtain the necessary certificates, you must have an active Apple Developer Program membership, which is $99/year.
> 2. Create Certificates in Apple Developer Account
> You’ll need to create a Developer ID Application certificate from your developer account. This certificate lets you sign and distribute apps outside the Mac App Store but with Apple verification, so users can trust the app.
> Optionally, you may also want a Developer ID Installer certificate, which is used to sign your app installer if you distribute a package installer (.pkg).
> 3. Code Sign Your App
> You can use Xcode or command-line tools to sign your app. The basic command for signing an app in the terminal is:
> bash
> codesign --sign "Developer ID Application: Your Name (Team ID)" /path/to/YourApp.app
> This command will apply the Developer ID Application certificate to your app bundle.
> To verify the app is signed, you can run:
> bash
> codesign --verify --verbose /path/to/YourApp.app
> 4. Notarize Your App (Optional but Recommended)
> While signing is essential, macOS also requires apps distributed outside the Mac App Store to be notarized for optimal compatibility with Gatekeeper. Notarization is an automated process by Apple that scans the app for malicious content and verifies it is signed correctly.
> Submit the signed app to Apple’s notarization service using Xcode or the command line:
> bash
> xcrun altool --notarize-app -f YourApp.zip --primary-bundle-id "com.yourcompany.yourapp" -u "your-apple-id" -p "app-specific-password"
> Once notarized, you’ll receive an email from Apple confirming the notarization, and you can staple the notarization ticket to the app:
> bash
> xcrun stapler staple /path/to/YourApp.app
> 5. Distribute Your App
> With code signing and notarization complete, your app is ready for distribution. Users will be able to install and run it on macOS without triggering security warnings from Gatekeeper.

Best of all would be to get OpenCPN into the MacOS app store.  It would be automatically signed and this would increase its visibility enormously.  But there may be difficulties meeting Apple's requirements re the user interface and sandboxing requirements.

> Getting an app into the macOS App Store involves meeting technical, content, and legal requirements to ensure quality, functionality, security, and adherence to Apple’s guidelines. Here’s an overview of the requirements:
> 
> 1. Developer Account
> You’ll need to enroll in the Apple Developer Program, which costs $99 per year. This allows you to access development tools, testing resources, and the App Store submission process.
> 2. Technical Requirements
> Sandboxing: Apps must be sandboxed to enhance security by limiting access to user data and system resources.
> Code Signing: Code signing is mandatory for all submitted apps, which certifies that the app is from a verified source and hasn’t been tampered with.
> 64-Bit Architecture: macOS apps must be built for 64-bit architecture to be compatible with recent macOS versions.
> Privacy Requirements: If your app accesses user data (like location, photos, microphone), it must include privacy usage descriptions in the app’s Info.plist file explaining why these permissions are needed.
> Compatibility: Your app should be compatible with the latest version of macOS, typically the most recent macOS release or two prior.
> 3. App Content and Functionality
> UI Standards: Apple emphasizes a high-quality user interface (UI). Your app should have a polished and user-friendly design that follows Apple’s Human Interface Guidelines (HIG).
[... cut ...]
> Data Privacy: Apps need to comply with data privacy laws like GDPR (Europe) or CCPA (California), especially if collecting personal data from users.
> 5. App Submission and Review Process
> App Store Connect: Submit your app using App Store Connect, where you provide metadata (app name, description, keywords), screenshots, and any promotional assets.
> App Review: Apple’s review process typically takes 1–3 days, although it may vary. The review team may request revisions to your app if it doesn’t meet the guidelines.

bdbcat, 2024-10-28:
Tony...
Very thorough synopsis.  Thanks.

antipole2, 2024-11-01:
Having just installed Sequoia, I have looked further at this but not yet actually tried to install OpenCPN.  My existing installation runs OK as it was installed before the update to Sequoia.

For the ordinary user, who mostly only installs each OCPN release once, Method 1 above should not be too onerous.  But we must ensure that users are forewarned and given the instructions before they can download OCPN.  Otherwise, we will have endless posts on the forum complaining they cannot install on their Mac.

For developers frequently installing test releases, this is a more significant inconvenience.  But they should be up to running the one-liner in Terminal after each install:
`sudo xattr -rd com.apple.quarantine /path/to/app.app`
Else they could _temporarily_ use Method 3 but I think I would stick with the above command.

bcn2, 2024-11-01:
My install was in parallel - a 5.11. In that case one is running into the new fence.
I do see indeed a big hurdle for a new "innocent" user.

Even if being present in the Apple AppStore would be perhaps ideal, starting to "notarize" OpenCPN for sideloading would already help a lot.

Questions coming into my mind:
- how does this integrate into the Cloudsmith workflow (or doesn't it fit)?.
- are there parts of the code or bundled libs that have to be modified?
- what happens with the plug-ins? Can those be installed without being signed and notarized?
- Betas (if not signed explicitly for official beta testing for example) or daily builds are not covered, what should not be an issue

To answer the question we will need somebody to test simply the actual OCPN version going through this process. Without hurry.

nohal, 2024-11-01:
For all this we need someone to pay to Apple. I used to have the subscription years ago and tested signing OpenCPN back then. There was no problem.

bcn2, 2024-11-03:
https://github.com/OpenCPN/OpenCPN/commit/e4988a36caf736b2d6febf2debb58d018f239a60
does install directly without queries from Gatekeeper and does run without issues at a first view.
Great....

cpelon, 2024-12-23:
I would consider maintaining a version of OpenCPN for macOS in the Mac Store. To cover the ongoing costs of this effort, would it be acceptable to charge a modest fee ($10-20) for the app? Would like to get thoughts on whether this approach would align with community expectations and licensing.

nohal, 2024-12-23:
Signing the application bundle is already implemented and the extortion fee for the certificate has been paid already. The signed package will be available starting with the next maintenance release.
I believe nothing changed on the fact AppStore is not compatible with GPL license. I would also say that the suggested fee is somewhat steep unless you expect major code contributions from your side will be required to make it possible.

bcn2, 2024-12-23:
What would we improve for OpenCPN (user) charging somebody at the AppStore? For a software you can get as well for free?
Apple yes, will take their 30%
Plus Nohal's observation about GPL and Apple conditions of service. What does bar OCPN for iOS as well.

nohal, 2025-06-02:
Yes, every release srating with 5.11 beta cycle is signed and notarized, currently https://github.com/OpenCPN/OpenCPN/releases/tag/v5.11.1-beta1