OpenCPN/OpenCPN #4187: Introduce a global section of corrections and damping in sensor connections

bc38, 2024-10-27:
When adding various sensors to CPN, I encountered the inability to enter corrections and adjust attenuation.
In the connection section, a section of corrections and attenuation is needed, all parameter changes should be applied globally to CPN and Plugins.
List of configurable parameters of prime importance:
SOG
COG
Heading
Depth
Water Temp
Apperent Wind Speed
Air Temp

transmitterdan, 2024-10-27:
This has been discussed several times in the past.  All commercial sensors, except some GNSS units, have offset and damping settings. We have generally held to the view that the sensor should be already corrected by the time the data arrives at OpenCPN. If it were not so, then there is the possibility that other displays would not match what OpenCPN shows and cause confusion.

We provide damping and magnetic correction for GNSS data because many units do not provide those features.

I think no matter what we offered in the way of "corrections" for other types of sailing instruments it would not make everyone happy,

bc38, 2024-10-27:
First of all, what I encountered was the Heading sensor course correction and water temperature.
You say that commercial sensors have a setting function? Which ones? Chartplotters and MFDs do, but then what's the point of CPN?
In my case, I use GNSS Lowrance Pint-1, DST-800, and even any other similar sensor does not have a correction setting without an MFD.

transmitterdan, 2024-10-27:
I am not aware of any MFD that applies the corrections itself.  All the ones I have seen simply program the sensor and then report what the sensor puts out.

transmitterdan, 2024-10-27:
> In any case, CPN is currently unable to correct the heading error angle, and it is not possible to correct the water temperature. Apparently, CPN is used as a backup to the MFD and not the main navigation system...

I'm not sure I agree with that.  OpenCPN is a fine navigation program assuming instruments output accurate data. It does many things much better than the typical MFD in my opinion.

The issue is whether or not OpenCPN should have the ability to download setup data into proprietary hardware such as heading/temp/depth sensors so that they will produce accurate data. OpenCPN is not proprietary but instead is open source so asking it to have proprietary coding is probably pointless.

Perhaps the question should be directed at the makers of inaccurate sensors as to why they do not provide a software tool to "calibrate" their sensors. But you know the answer to that.

TwoCanPlugIn, 2024-10-27:
I personally agree with Dan. (I was originally typing something that was not fit to be printed).

refer to the following thread.

[https://www.cruisersforum.com/forums/f134/how-to-calibrate-stw-from-transducer-290188.html](url)

Analogue sensors are calibrated at the device to which they are connected. 

Smart sensors, be they NMEA 0183 or NMEA 2000, are generally calibrated at the sensor, usually by proprietary NMEA 0183 sentences or NMEA 2000 PGN's.

For example calibration of most modern Airmar sensors over NMEA 2000 requires a combination of PGN's 126208 (Command Group Function) and 126720 (proprietary multi frame messages).

Due to the proprietary nature of the calibration protocols, calibration can generally only be performed using the instrument displays or multi-function displays from the vendors who oem the sensors.