OpenCPN/OpenCPN #4197: comm_drv_n0183_net: Broadcast address problems

leamas, 2024-10-31:
**Describe the bug**
The  N0183 network driver tries to figure out if an address actually is a broadcast address. This logic does not work for netmasks with more than 24 bits for example a /28 sliced network

**To Reproduce**
Steps to reproduce the behavior:
1. Check comm_drv_n0183_net.cpp around line  220, find this snippet:
```
    if ((!m_is_multicast) && (m_addr.IPAddress().EndsWith("255"))) {
      int broadcastEnable = 1;
      m_tsock->SetOption(SOL_SOCKET, SO_BROADCAST, &broadcastEnable,
                         sizeof(broadcastEnable));
    }
```
2. Note that a broadcast address on a /28 network does not (necessarily) ends with 255.

**Expected behavior**
The logic should take the netmask into consideration, the address is a broadcast address iff all bits outside the netmask are set.

**Additional context**
Since almost all users, also developers, uses private ipv4 class C addresses this is not reported in  the  wild. It's still a bug, though.

TwoCanPlugIn, 2024-10-31:
This implies that you would need iterate through a list of network connections and retrieve each netmask. 

Does this mean you're going to reopen this can of [worms](https://github.com/OpenCPN/OpenCPN/issues/3700)?

All I can add is "good luck"

leamas, 2024-10-31:
I'm not saying or how this should be fixed. Just documenting something I found when working with the code. 

Might very well end up in "known bugs"

TwoCanPlugIn, 2024-11-02:
More of an FYI, or interesting coincidence, and as you said, very unlikely to occur on an OpenCPN users's boat network.

I just happen to be in an apartment where the WiFi Access Point has assigned my laptop with the IP Address:
192.168.68.78 and a network mask of 255.255.252.0

Most non-network savvy users would think their broadcast address is 192.168.68.255, however the correct broadcast address is actually 192.168.71.255

Whilst the current behaviour of OpenCPN, would detect the trailing ".255", and set the SO_BROADCAST flag correctly, there is a possibility that other devices may have been allocated any addresses in the range  192.168.68.1 - 192.168.71.254. In that scenario any data sent to the "incorrect" broadcast address of 192.168.68.255 may never reach other devices.

This is an example where good UI design and automated calculation of the broadcast address could eliminate configuration errors.

leamas, 2024-11-02:
However, calculating the netmask means using  native code which is a pain and certainly will take a while. The only reasonably quick fix I can think of is  some kind of "Advanced..." thing which opens up the possibility to set the netmask.

After all, lot's of consumer hardware requires user to set router, address and netmask. It should no be that bad.

And then again, ipv6 solves all of this. It isn't **that** far-fetched, the auto-configuration capabilities of ipv6 suits a yacht network quite well. It's no coincidence that  OneNet runs on ipv6

TwoCanPlugIn, 2024-11-03:
Calculating the Broadcast Address is generic code, it is the retrieval of the network interfaces and their IP Addresses and Network Mask which require native code.

However, similar to USB serial device insertion/removal, you've probably forgotten that you already use Linux native code to populate the list of socketcan interfaces. Just a slight reworking of that code and the inclusion of the Windows and Mac OSX equivalent.

Find attached a very quick & dirty proof of concept for Windows/Linux. Should be very easy for you guys to sort out. I shan't comment on the UI design, don't want to get my head bitten off again.

[adapterinfo.txt](https://github.com/user-attachments/files/17610397/adapterinfo.txt)

leamas, 2025-02-23:
Looking back at this the new communications dialog has an "Advanced" panel which seems to  natural place for a "Set netmask" option.  The whole thing is hidden by default, and while the broadcast address is obscure it is  not worse than other things here.

For users, setting the netmask is better than setting the broadcast address -- the netmask is commonly  used when setting up various consumer equipment, the broadcast address is not.

Yes, digging up the netmask from the drivers is a better solution. However, I consider this to be too much work for this bug.  Nothing stops us from expanding a manual solution with better default values in a later step.

leamas, 2025-02-25:
A worse problem is that PR hack adds the netmask as a global setting. While this should work for almost all live setups, the correct solution would be to bind the setting to the actual interface.  This would handle the  use case where an opencpn user have multiple network connections, a use case yet to be seen...

leamas, 2025-02-25:
OTOH, looking at this with fresh eyes: Perhaps this would be better:

[image]

@TwoCanPlugIn Thoughts?

TwoCanPlugIn, 2025-02-25:
OK, displaying the resultant dotted notation as the user selects the number of netmask bits looks a little bit more intuitive, as they will have seen the dotted notation in the network dialogs for their respective OS, WiFi Access Points (including those on NMEA 0183 & 2000 gateways).

Re, the setting, I had assumed it was for each individual connection, rather than global. As you say probably an edge case and probably only one that developers face, but then again, isn't PR all about solving an edge case?

Nonetheless, you know what my preferred solution is:-)

leamas, 2025-02-25:
> Nonetheless, you know what my preferred solution is:-)

Patches welcome ;)

leamas, 2025-02-25:
Closed by e0e1d6d

closed 2025-02-25 by leamas