OpenCPN/OpenCPN #4200: No OpenGL mode: Tide symbols overlap Toolbar

Hakansv, 2024-11-01:
**Describe the bug**
When in no OpenGl mode the tide symbols is shown over the toolbar. See shot.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable OpenGl
2. Enable "Show tide stations"
3. Locate the chart canvas so some tide stations is close to the left screen edge
4. Be sure the toolbar is shown

**Expected behavior**
The tool bar should be in foreground as in OpenGl mode

**Desktop (please complete the following information if applicable):**
 - OS: Win 10/11
 - Version: Built on today's master (Last commit, although not the cause: 0e8e37b63)

**Screenshots**
[image]

leamas, 2025-05-02:
Using current master at 47ba7964d I cannot reproduce this, neither on Linux nor Windows

@Hakansv : Can you?

Hakansv, 2025-05-02:
No, it's gone now. Sorry, I haven't checked for while so no idea about the fix.
Many thanks for cleaning up, appreciated!

closed 2025-05-02 by Hakansv