OpenCPN/OpenCPN #4201: Saving navobj.xml every 24h

bcn2, 2024-11-02:
We have been confronted right now with the case of a sailor who crashed OpenCPN when it was running since 48 hours.
With the loss of the actual navobj.xml

The back-up strategy has already improved a lot with 5.10

An automatic back-up every 24h for systems on longer terms would round this up further more.
Alternative: a user defined period - "back-up every X hours"

TwoCanPlugIn, 2024-11-02:
Perhaps that's why databases with "ACID" characteristics were invented.

I've never understood why OpenCPN has never used SQLite to store waypoints, routes and tracks especially as it is one of the included libraries.

bcn2, 2024-11-02:
Perhaps as the concept is from about 2009 when GPX was the only existing standard?

TwoCanPlugIn, 2024-11-02:
GPX is simply a XML Schema that provides a standard format for the exchange of GPS data such as tracks, waypoints and routes.
.
It doesn't define how the underlying data is stored on disk. 

A database such as SQLite which has the concepts of transactions, journals, and logs is always going to be much more resilient than just writing text to a file.

gibbycu, 2024-12-19:
I too had this happen losing 12hrs worth of track.  In the interim can the navobj.xml.changes file be copied and saved to a renamed file after a crash instead of just being deleted. At least then we have the possibility of extracting the useful data after a crash even if OpenCPN cannot load it.

nocodehummel, 2025-01-09:
Just lost my planning work after a crash caused by a plugin. I agree that a proper database solution is needed to prevent database. A workaround (for now) could be to recover the CHANGES XML file when the user selects to start in save-mode.

bsteubing, 2025-04-27:
OpenCPN crashes about once every day to two days on my computer. I keep loosing tracks and waypoints. When I re-open the navobj file seems to be overwritten, so it is permanent data loss. I am not familiar with the OpenCPN codebase, but I don't understand why:
- track and other data isn't simply written to a file as it is created
- or if it is, why re-starting OpenCPN after a crash does not retain this data
- why it is limited to saving every 24h; I would personally save every 30 min or so, just to avoid loosing tracks so often (I understand it is hard to elimiate all crashes, but please try to find solutions to minimize data loss - I have been using OpenCPN for 10 years now and I am still a big fan, but it is simply worse now than it used to be in this sense...)

bdbcat, 2025-04-27:
In the spirit of complete transparency, it will be helpful to describe the current model.

1.  Small (or large) changes to the navobj internal store are not written to the disk file as XML immediately.  Why?  Because of the resulting performance hit.  The navobj.xml file can regularly grow to multi-megabyte size.  We would not want to recreate the entire file just to change the name of a waypoint, for example.
2. What is done is that all incremental changes are written to the (much smaller) changes file, as the changes occur.
3. What is designed to happen is this:  On a restart after a crash, the system detects a not-clean shutdown, finds a changes file, and applies those changes to the most recent navobj.xml file.  It then deletes the changes file.  Theoretically, the state is as though the system was shutdown cleanly.

The saving of navobj.xml on a fixed schedule does this:  If no changes detected in the navobj data, do nothing.  If there is a live changes file, then apply the changes file to the most recent navobj.xml file, delete the changes file, reload the navobj file, and carry on.

So, we can say simply, that if the above process does not work, then there is yet some bug in the process.

This is extremely difficult to capture alive, and study.  Why?  Because the normal (and reasonable) user reaction to a crash is to restart OCPN immediately.  Thus the crash state is lost, and the exact code paths involving application of changes file to navobj.xml is also lost.
Further, if there seems to be a problem on changes application. the user will naturally restart OCPN, again.  More state info lost.  No possibility to debug.

We ask for users to "full stop" when this happens in real life, and let us see the files to prove that the changes application will do the right thing.  Rarely do we get useful data.

And that is where we are.

I think the idea of increasing the flush/reload frequency is probably our best path forward at the moment.  I suppose a ridiculous value of once every ten minutes would work, provided we can arrange to not collide with any data editing actions asynchronously.  Have to think about that.  For example, simple dragging a waypoint induces an addition to the changes file.  We may actually need to freeze the GUI during the flush/reload cycle.  More thought required.

Thoughts?

bdbcat, 2025-04-27:
And, of course, it goes without saying (or does it?) that successful recovery from a crash is far more likely on the most recent OpenCPN production version, whatever that may be.

bdbcat, 2025-04-27:
"A database such as SQLite which has the concepts of transactions, journals, and logs is always going to be much more resilient than just writing text to a file."

Totally agree.  We would welcome seeing a skilled dev take this one on.  So far, no takers.  So it does not look like it will make it into coming o5.12 Release.

bsteubing, 2025-04-28:
and if you need me to test anything, just let me know anytime

bdbcat, 2025-05-01:
bsteubing...
Thanks for the comments.
I am currently working on an update to the track management logic.
If you have a navobj.xml file with lots of tracks, and lots of trackpoints, it would be very helpful for stress testing.
Thanks
Dave

bdbcat, 2025-05-03:
I have pushed a commit related to this bug/issue.

This patch updates the navobj persistence strategy to store tracks and trackpoints in an SQLite database. Details follow.

1.  This patch is transparent to users, no config changes required.
2.  The track database is stored in the config directory as "navobj.db".
3.  The existing "changes file" logic is not used for track database updating.  All tracks and trackpoints are immediately written to the database as created.
4.  On first run, after failing to find file "navobj.db", the following steps are performed:
    a.  Make a verbatim safety backup copy of navob.xml, called navobj.xml.backup.
    b.  Create the initial empty "navobj.db" file.
    c.  Import existing tracks from navobj.xml, inserting into the new database immediately.
    d.  Remove all tracks imported into the database from navobj.xml.  This file will be updated on on next app shutdown.
    e.  navobj.xml.backup will not be automatically removed.  It's creation date will serve as a checkpoint for reference.
    f.  Result: a new navobj.db file, and a (probably) smaller navobj.xml file.
Notes:
1.  As you can see by (3), the integrity of track data is greatly improved. The track database is always up-to-date, consistent, and highly resistant to unrelated faults within OCPN core.
2.  The longer term plan is to migrate the Routes and Waypoints to the same navobj.db database.  At that point, navobj.xml, and all code paths related, will disappear. Schedule for this update is undetermined.

bcn2, 2025-05-03:
Tracks with the highest priority as they trigger the most frequent single and unattendent updates of the data. Hence the biggest risk to be trapped by some crash without some former back-up, ending up with inconsistencies. And data loss.

rgleason, 2025-05-03:
@bsteubing  [Dave merged into the master branch](https://github.com/OpenCPN/OpenCPN/commit/84f2b70bf9e5d825a27d753a6fc4464c45b2dc04)  can you build and test?

bcn2, 2025-05-03:
On macOS 15.4.1:

[image]

bcn2, 2025-05-03:
> [@bsteubing](https://github.com/bsteubing) [Dave merged into the master branch](https://github.com/OpenCPN/OpenCPN/commit/84f2b70bf9e5d825a27d753a6fc4464c45b2dc04) can you build and test?

Available on Cloudsmith   ..5.11.0  84f2b70..
https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/?page=1

bsteubing, 2025-06-06:
> Available on Cloudsmith ..5.11.0 [84f2b70](https://github.com/OpenCPN/OpenCPN/commit/84f2b70bf9e5d825a27d753a6fc4464c45b2dc04).. [...]

Thanks a lot for this link. I cannot build it myself, but via cloudsmith I am able to download the setup .exe file. Sorry for the silence, I have not been able to test for the last month, but I have just installed the [latest version](https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/detail/raw/opencpn_5.11.0-0%252B3306.05585e8_setup.exe/-/) of OpenCPN from cloudsmith for Windows. I installed it on 2 computers and it runs fine for now. As intended, the navobj.xml is gone the second time I open OpenCPN and the track seems to be in the navobj.db

THANK YOU for this. Implementing an SQlite solution seems like the right pathway forward and will hopefully reduce the chance of data loss with a crash (regardless of this is caused by OpenCPN or graphics driver or ...)

closed 2025-06-09 by bcn2