OpenCPN/OpenCPN #4212: Grib plugin - symbol sizing on Retina displays (OpenGL)

nohal, 2024-11-14:
The wind barbs and wave symbols are too small on Retina displays, the usual x2 scaling needs to be applied to them.

nohal, 2024-11-15:
AFAIK all Retina displays report half it's actual physical resolution. But I certainly only have access to Macbooks, on which there also probably is very little if any reason to change the native resolution.

sebastien-rosset, 2024-11-15:
As part of investigating #4211, I found out on my Mac, the scale factors (1 or 2) depends on the selected resolution.

nohal, 2024-11-15:
This actually gets even more interesting with multiple monitors with very different resolutions, on different platforms and when at least some reasonable results are required while using wx3.0. As wxWidgets is not excelling in this area, we collect the actual monitor information in https://github.com/OpenCPN/OpenCPN/blob/master/gui/src/displays.cpp and should provide it also to the plugins through the API.