OpenCPN/OpenCPN #4213: Provide po files for `en_US` localization

sebastien-rosset, 2024-11-15:
**Is your feature request related to a problem? Please describe.**

The `.po` files currently provide localized content for `en_GB`. There is no variant for `en_US`.

**Describe the solution you'd like**

Provide po files for `en_US`.

**Describe alternatives you've considered**

No change, and the en_GB content is displayed.

**Additional context**

I'm willing to provide the content if this proposal is accepted.

nohal, 2024-11-15:
I can enable `en_US` on Crowdin with more or less one click, but given it is the actual source language used in the sources, is there really a reason for it? Do you see some terms that need additional translation between the sources and `en_US`?

sebastien-rosset, 2024-11-15:
> There are just a couple of translatable strings containing "colour" in the whole codebase (7 in total). Probably better to fix them than treating `en_US` as a translation.

Sure, I can raise a PR with these changes.

PR #4214 (2024-12-29): Change translatable strings to use US English instead of British English
Standardize the spelling of British English words ("color" and "licence") to US English for translatable strings in the source code.

Fix #4213
closed 2024-12-29 by leamas