OpenCPN/OpenCPN #4215: OpenCPN crashes with segmentation fault while opening SHP file

sebastien-rosset, 2024-11-20:
**Describe the bug**

OpenCPN crashes with segmentation fault on startup after displaying message "OpenCPN needs to update the chart database from config file entries..."

**To Reproduce**
Steps to reproduce the behavior:
I'm not sure exactly how to create a reproducible case. However, from the code it's obvious why the code is crashing.

**Expected behavior**
OpenCPN should not crash with segmentation fault.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: MacOS
 - Version Sequoia 15.1

**Additional context**

The problem is in LoadSHP:
```c++
bool ShapeBaseChart::LoadSHP() {
  _reader = new shp::ShapefileReader(_filename);
  auto bounds = _reader->getBounds();   <== This is where the code crashes.
  if (!_reader->open()) {
    return false;
  }
```

However, there is no guarantee the reader will be opened, hence `dbf` may be null
```c++
    ShapefileReader::ShapefileReader(std::string fileName) {
        shp = SHPOpen(fileName.c_str(), "rb");
        size_t i = fileName.find(".shp");
        dbf = DBFOpen(fileName.replace(i, 4, ".dbf").c_str(), "rb");
    }
```

Suggested change is to check whether the reader is opened or not:
```c+=
  _reader = new shp::ShapefileReader(_filename);
  if (!_reader->open()) {
    return false;
  }
```

bdbcat, 2024-11-21:
Looks good.  Bring a PR, I'll accept.

sebastien-rosset, 2024-11-21:
> Looks good. Bring a PR, I'll accept.

I've raised PR #4216

closed 2024-11-26 by sebastien-rosset