OpenCPN/OpenCPN #4217: Use of multiple CM93ed2 directories not possible in chart files & chart groups

Gebruikers-naam, 2024-11-20:
I use 4 separate chart groups for my 4 different directory version of CM93ed2. Plus more chart groups for USA or NZ charts and mbtiles.
I upgraded from 5.6.2 to 5.10.2 and this is not possible anymore. 

To Reproduce:
It can go wrong in various ways. For instance: I create charts groups for my 4 CM93ed2 directories, and add these directories to the respective chart groups, and click OK. And I get an immediate popup warning that one of the CM93ed2 chart groups is empty. They are then all 4 empty except for one.
Or, if the popup warning does not happen: Then I try to switch between CM93ed2 chart groups by clicking on the 1....4 keys. Then again I get these popups warnings telling me that CM93 chart groups are empty. When I check that chart group, initially the directory might still be shown in that chart group, but by for instance clicking "apply" it disappears. 
Ultimately and consistently, all 4 CM93ed2 chart groups are empty, except for one. 
This all while doing sufficient database rebuilds between the various actions and clicking "apply".
I then downgraded to 5.8.4. Same problem. I downgraded further to 5.6.2 and the problem disappeared.

Screenshots:
Video screenshot of 100 MB showing the problem can be downloaded via 
https://www.dropbox.com/scl/fi/fhn0z5otd35y0xgfmir91/openCPN-bug-report-chart-groups-2024-11-20-105633.mp4?rlkey=7m083vphl8lmclx57ch7903iy&dl=0

Desktop:
- W11 pro on 64 bit AMD and W10 on Intel
- openCPN 5.8.4 and 5.10.2. It works correctly on 5.6.2, tested by downgrading from 5.10.2 to 5.8.4 to 5.6.2

Additional context
Having multiple different CM93ed2 chart directories is useful, in 3rd world countries the detail level between versions is quite different

closed 2025-03-22 by bdbcat

Gebruikers-naam, 2025-03-23:
Great. Will this be in next release?

bdbcat, 2025-03-23:
Yes, it will be.
If you would like to test, you may try this pre-Beta installer for Windows.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.11.0-0+2848.2f8f2d1_setup.exe:

commit 2f8f2d1c (2025-03-22): Allow multiple instances of cm93 chartsets in database.  Closes #4217