OpenCPN/OpenCPN #4218: zoom in is not properly centered when the scale factor != 1 and OpenGL is disabled

sebastien-rosset, 2024-11-21:
**Describe the bug**
The center of the chart shifts to the South East after zooming in. This occurs specifically when OpenGL is disabled, and the screen scale factor is not `1`.

**To Reproduce**
Steps to reproduce the behavior:
1. Set the screen resolution such that the scale factor is not set to 1. For example, on a macbook pro, the scale factor is often set to 2.
2. Launch OpenCPN.
3. Go to settings -> Display -> Advanced and disable OpenGL. Click OK
4. Click `+` to zoom in. The chart is zoomed in.
5. Notice the center of the chart has been shifted to the South East of the previous zoom level.

**Expected behavior**
After zooming in, the center of the chart should be the same as what it was at the previous zoom level.

**Screenshots**
See attached video.

**Desktop (please complete the following information if applicable):**
 - OS: macOS
 - Version Sequoia 1.15.

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

sebastien-rosset, 2024-11-21:
The problem seems more serious. I'm not sure if I'm doing something wrong.
When OpenGL is enabled, centering the viewport at lat=30, lon=30.
When OpenGL is disabled, the lat/long coordinates are wrong.

https://github.com/user-attachments/assets/aca96a48-643a-4dcd-874f-8cd9854e17ea

sebastien-rosset, 2024-11-21:
Gibraltar with OpenGL enabled is at 35.5N,5.3W
[image]

Gibraltar with OpenGL disabled: the grid showing coordinates is the same as the above image. However, the lat/long coordinates at the cursor are wrong: 31.03N, 0.55E (as shown in the bottom status bar).

[image]

nohal, 2024-11-21:
AFAICT all these problems have a common cause - on Retina displays the chart canvas is twice the size it should be in non-OpenGL mode and the screen actually shows only it's top left quarter.

sebastien-rosset, 2024-11-21:
> AFAICT all these problems have a common cause - on Retina displays the chart canvas is twice the size it should be in non-OpenGL mode and the screen actually shows only it's top left quarter.

I'm debugging `ViewPort::GetLLFromPix` to understand the specific differences.
```
# without OpenGL
09:26:16.874 MESSAGE viewport.cpp:340 pix_width: 2862, pix_height: 1560, dx: 17.000000, dy: -372.000000, type: 1, lat: 32.656791, lon: -0.552489, p.m_x: 1448, p.m_y: 1152
09:26:16.898 MESSAGE viewport.cpp:340 pix_width: 2862, pix_height: 1560, dx: 17.000000, dy: -372.000000, type: 1, lat: 32.656791, lon: -0.552489, p.m_x: 1448, p.m_y: 1152
09:26:16.898 MESSAGE viewport.cpp:340 pix_width: 2862, pix_height: 1560, dx: 17.000000, dy: -372.000000, type: 1, lat: 32.656791, lon: -0.552489, p.m_x: 1448, p.m_y: 1152

# with OpenGL
09:25:04.857 MESSAGE viewport.cpp:340 pix_width: 2862, pix_height: 1560, dx: 53.000000, dy: -268.000000, type: 1, lat: 35.980343, lon: -5.563329, p.m_x: 1484, p.m_y: 1048
09:25:05.158 MESSAGE viewport.cpp:340 pix_width: 2862, pix_height: 1560, dx: 53.000000, dy: -268.000000, type: 1, lat: 35.980343, lon: -5.563329, p.m_x: 1484, p.m_y: 1048
09:25:05.170 MESSAGE viewport.cpp:340 pix_width: 2862, pix_height: 1560, dx: 53.000000, dy: -268.000000, type: 1, lat: 35.980343, lon: -5.563329, p.m_x: 1484, p.m_y: 1048
```

sebastien-rosset, 2024-11-22:
Perhaps a related problem is that when the user toggles between OpenGL and no OpenGL, the chart scale changes. IMHO, that should not be the case. The chart scale should remain the same.

sebastien-rosset, 2024-11-22:
> AFAICT all these problems have a common cause - on Retina displays the chart canvas is twice the size it should be in non-OpenGL mode and the screen actually shows only it's top left quarter.

I think I understand what you meant. Given a screen dimension, in "OpenGL disabled" mode, the displayed chart is 1/4th the area of the chart in "OpenGL enabled" mode. The region of the chart being displayed in the top left corner of what is displayed with OpenGL enabled.

I took a screenshot with/without OpenGL and scaled down the "OpenGL disabled" screenshot such that it aligns with the chart scale when OpenGL is enabled.

[image]

bdbcat, 2024-11-22:
I understand the desire to be correct in all modes, and I support that, but...
Is there any possible scenario where a Mac user would _NOT_ want to use GL?
Just asking....

sebastien-rosset, 2024-11-23:
> I understand the desire to be correct in all modes, and I support that, but... Is there any possible scenario where a Mac user would _NOT_ want to use GL? Just asking....

Initially, I hit this problem after building and running locally. When running `cd build ; ./OpenCPN.app/Contents/MacOS/OpenCPN` then OpenGL is disabled and the option to enable is greyed out. Later I found out how to run with OpenGL. I agree that I will always run with OpenGL, and nonetheless the software should without this kind of issue. For me it's an opportunity to understand and document the code, hopefully to help contributors including myself. Several functions/fields have no trivial semantic solely based on the names. But let me know if you don't want this, I will abandon.

nohal, 2024-11-23:
Even if we decided to completely remove non-GL mode on macOS, the work done here sure makes sense as we do want to revisit the scaling on all the platforms and improve HiDPI and multi-monitor behaviour.

bdbcat, 2024-11-23:
Yes, I agree.  Good to continue,  Just wondered what motivated this task.

sebastien-rosset, 2024-11-26:
I've root caused the problem in `wxPoint ViewPort::GetPixFromLL()`:

https://github.com/OpenCPN/OpenCPN/pull/4219/files#diff-873556a1e246cee5e7cb51d0de3e961fad061365ed62e18b59e847aa2a366557R262-R268

There is a 9-year old comment that states
```c++
// TODO: eliminate the use of this function
wxPoint ViewPort::GetPixFromLL(double lat, double lon) {
```

1. The mouse pointer coordinates that are displayed in the status bar are correct (previously they were wrong).
2. The center coordinates of the chart are now the same as in OpenGL mode.
3. The coordinates of the chart corners are now the same as in OpenGL mode.
4. The area covered by the chart is the same as in OpenGL mode.
5. The chart scale does not change (i.e. the chart scale remains the same when switching between OpenGL and no-OpenGL mode).
6. The physical length on the screen matches the displayed chart scale. For example, 1000 nautical miles at 1:13.2M scale should be 140.3 millimeters on the screen.

Now I need to find a way to replace the `2` constant with a variable

sebastien-rosset, 2025-03-29:
#4219

closed 2025-03-29 by sebastien-rosset