OpenCPN/OpenCPN #4220: Appearance of tracks changes at smaller scales

Gebruikers-naam, 2024-11-21:
**Describe the bug**
When I change the scale of the chart, the appearance of tracks changes. 

**To Reproduce**
Steps to reproduce the behavior:
Take a locations where there are multiple tracks. Change some of them in colour, style (for instance dashed) and width.
Change the scale; in my case from 1:1.2M to 1:564000
See the appearance (colour, width, style) of the tracks on the screen change.

**Screenshots**
You can download a video of this behaviour from 
https://www.dropbox.com/scl/fi/ojkf8oczz03j7zapc90ip/Track-appearance-changing-at-different-scales-2024-11-21-143152.mp4?rlkey=6cns3iaypxoc5ep11bq9wvjsh&dl=0

**Desktop (please complete the following information if applicable):**
 OpenCPN 5.10.2 on 2 PC's, both with openGL:
W11 Pro on 64 bit AMD
W10 on 64 bit Intel i5

nohal, 2024-11-21:
Not a bug, disable track highlighting on the Ship->Own ship tab in the Toolbox if you don't like the behavior.

bdbcat, 2025-03-17:
Closing as resolved

closed 2025-03-17 by bdbcat