OpenCPN/OpenCPN #4223: Can't scan Bluetooth devices on Android

Rodemfr, 2024-11-23:
**Describe the bug**
When creating a new Serial/Bluetooth connection, the "BT Scan" button does nothing making impossible to connect to a new BT device.

**To Reproduce**
1. Go to Options->Connection
2. Create a bluetooth serial connection
3. Click "BT Scan" button while your BT sevice is active and already paired to your tablet/smartphone
4. OCPN does not list it in the device list

**Expected behavior**
See the BT device in the list of available devices

**Screenshots**
N/A

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Bug observed on Android 8 (Galaxy Tab A6) and Android 12 (Xiaomi Smartphone)

**Additional context**
Not that it is not entirely impossible to connect to a BT device because it is still possible to manually edit the config file and to add the MAC address and the name of the device with a text editor.

bdbcat, 2024-11-23:
Have you upgraded OpenCPN for Android to current Release: 5.10.2{114} ?

Rodemfr, 2024-11-24:
No I was on 5.10.0. It is now working with 5.10.2. Thanks for the help and sorry for the noise...

closed 2024-11-24 by Rodemfr