OpenCPN/OpenCPN #4227: wxCurl corrupts binary files when transferring via ftp

deks-awash, 2024-11-26:
**Describe the bug**
OpenCPN 5.8.4. The weatherfax plugin uses wxCurlDownloadDialog to download images from the internet. Seemingly the only site using ftp in WeatherFaxInternetRetrievalis the Australian Bureau of Meteorology where gif images are downloaded from ftp.bom.gov.au, and these are corrupted. The dialog uses wxCurlFTP which is hard coded to use ASCII transfer mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the weatherfax plugin
2. Click on the weatherfax icon in the main toolbar
3. Select menu Retrieve > Internet
4. Under Servers choose "Australia BOM", Regions "Australia", then "MSLP 12hr Forecast" in main panel
5. Under "retrieve", click the "Selected" button
6. OpenCPN will either crash or display a very distorted fax on the map

**Expected behavior**
OpenCPN should not crash when downloading binary images from an ftp server, and downloaded binary images should not be corrupted

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: linux - Ubuntu 24.04 (lubuntu)
 - OS Version 24.04
 - OpenCpn version 5.8.4
 - weatherfax plugin version 1.9.111.1

**Smartphone/Tablet (please complete the following information if applicable):**
 - not tested 

**Additional context**
Finding the url of the file attempted to be downloaded as (n this case) ftp://ftp.bom.gov.au/anon/gen/difacs/IDX0131.gif and using curl to download it directly works fine - the gif can be displayed without issue. In lib/wxcurl/src.ftp.cpp (which the dialog apparently uses) the wxCurlFTP constructor sets m_tmMode(kASCII) and there seems to be no mechanism exposed upstream to change it. The simplest fix would seem to be to test the url ending for a number of known binary file types (e.g. gif, png - perhaps configurable?) and set the transfer mode to binary accordingly.

deks-awash, 2024-11-27:
OK, a correction - the bug is not in the wxcurl library in the OpenCPN repo library (well, it's replicated here, but not run). I found the opencpn-libs repo which also has the wxcurl library with the same issue, rebuilt the weatherfax_pi plugin library with mode changed to binary - and images download correctly from BOM ftp site. So - is this a bug with the opencpn_libs repo, or with the weatherfax_pi - or both?

rgleason, 2024-11-27:
@deks-awash  Thank you. To fix wxfax specific issue, can you make a PR to my repository? 
I assume the opencpn libs needs to be fixed too, but getting that done may be a problem as the programmers have a schedule for that. Pavel, Alec or David may comment on that.

deks-awash, 2024-11-30:
Per advice, recreated as opencpn-llibs issue 17, and closing here.

closed 2024-11-30 by deks-awash

rgleason, 2024-11-30:
This plan sounds good. Thanks very much.