OpenCPN/OpenCPN #4228: Multi-layer gribs

sebastien-rosset, 2024-11-27:
**Is your feature request related to a problem? Please describe.**

OpenCPN has a built-in grib plugin to load weather grib files. Users can download various grib files, such as GFS, HRRR, ECMWF, ROTFS. Grib files can be used to forecast the weather, display a weather overlay on top of existing charts. For example, users can display wind, waves, rain, etc. 
There is limited support for loading multiple grib files at the same time and managing the content of the grib files. For example, it's possible to load one grib file that has waves and another one that has wind, but loading two gribs having wind data is not well supported.

**Describe the solution you'd like**

It would be useful to have a way to load multiple grib files in memory, and have the ability to configure what attribute(s) to display for each grib file.

1. Ability to load multiple grib files in memory. For example, the user might load HRRR for the 48-hour forecast and GPS/ECMWF for the next 6 days.
2. The UI should expose "layers", one per grib file.
3. The UI should show which grib files have been loaded in each layer.
4. The UI should expose actions to enable/disable the layers.

**Describe alternatives you've considered**
User can load/unload grib files one after the other.

**Additional context**
In the model:
1. Support an ordered list of grib layers.
2. Each layer may have 0 or 1 grib file. 
3. Each layer may be enabled or disabled. 
4. Add public methods to get weather data for a specific grib layer. 
5. Add public methods to get weather data across all layers. In that case, the implementation returns whether data with the best accuracy. If two layers have the same accuracy, then the data is prioritized according to the ordered layers in the list. 
6. Public methods should have arguments for specific coordinates and timestamp. For example, what is the wind forecast on March 3rd at 12:00 p.m. at specific latitude and longitude coordinates.

sebastien-rosset, 2025-01-20:
If anybody is interested in discussions, I've started a draft implementation at https://github.com/sebastien-rosset/OpenCPN/pull/4. This is not functional yet.

rgleason, 2025-01-20:
Sebasetien, this looks like a great beginning. 

Will have a "View" or "tabulation"  to improve our Meteo understanding of the weather_routing  "optimized" routing?  IE an image of the Grib data (state represented by color) , across time with color representation for waves, wind,  clouds, rain, etc. for that segment of time)? Wind would also have a direction component. And a row for port, starbooard, reach and downwind? along the optimized route?   Perhaps with the ability to select from several different routes?

rgleason, 2025-01-20:
Also I hope the PR for global control of time is part of this effort.

sebastien-rosset, 2025-01-20:
> Sebasetien, this looks like a great beginning.
>
> Will have a "View" or "tabulation"  to improve our Meteo understanding of the weather_routing  "optimized" routing?  IE an image of the Grib data [...]

For now I'm focusing on the grib data model:
1. create layers
2. enable/disable layers.
3. Interpolate data across layers, both spatially and temporally. Algorithms to calculate weather data for a given geographic area and time, considering the layered gribs may or may not have the same parameters (wind, gust, wave), precision...

I think the first step is to provide the right model and APIs to query weather data across layers. Then other aspects can be developed, such as the UI, integration with weather routing, tactics, etc.

rgleason, 2025-01-20:
Good approach.