OpenCPN/OpenCPN #4229: GridDraw is off (a little)

RooieDirk, 2024-11-27:
It seems GridDraw is drawing the grid lines a little off position. I made a mark # for double checking (attached)
[image]
Placing this mark over a known position in the chart is correct.
[image]

Placing the same mark on a gridline crossing show this:
[image]

Zooming on the position (52N, 5E)  will give slightly different results. The latitude grid is changing from 0 to 1 pixel off (below) an the longitude varies from 1 to 2 pixels to the right. 
Looks like a rounding issue somewhere to me.

Running on a debian12 box.

sebastien-rosset, 2024-11-27:
Just curious, what is the scale you are using?

RooieDirk, 2024-11-27:
Scale doesn't change anything. Always 1 to 2 pixels off. GL on/off doesn't help either.

sebastien-rosset, 2024-11-27:
As a side note, I found #4230 while looking at this issue.

sebastien-rosset, 2024-11-27:
I'm able to reproduce this issue with the following svg:
```svg
<svg xmlns="http://www.w3.org/2000/svg" viewBox="-50 -50 100 100" width="20mm" height="20mm">
    <!-- Horizontal lines -->
    <line x1="-40" y1="0" x2="40" y2="0" stroke="red" stroke-width="1"/>
    <line x1="-18" y1="-10" x2="18" y2="-10" stroke="red" stroke-width="1"/>
    <line x1="-18" y1="10" x2="18" y2="10" stroke="red" stroke-width="1"/>
    
    <!-- Vertical lines -->
    <line x1="0" y1="-40" x2="0" y2="40" stroke="red" stroke-width="1"/>
    <line x1="-10" y1="-18" x2="-10" y2="18" stroke="red" stroke-width="1"/>
    <line x1="10" y1="-18" x2="10" y2="18" stroke="red" stroke-width="1"/>
  </svg>
```

sebastien-rosset, 2024-12-06:
In addition, I've noticed there is a zero or one pixel misalignment between the major grid line and the minor grid ticks, both on the latitude and longitude grid.

[image]

sebastien-rosset, 2024-12-06:
I've noticed the following:
1. To draw the major grid lines, the `GetDoubleCanvasPointPix` function is used, which returns `double` values.
2. To draw the minor ticks, the `GetCanvasPointPix` function is used, which returns `int` values.

Internally, `GetCanvasPointPix` calls `GetDoubleCanvasPointPixVP`, then calls `wxRound`, which is a wrapper around [std::lround()](https://cplusplus.com/reference/cmath/lround/), round to nearest and cast to long integer.

The `DrawLine` function takes integer arguments:
```c++
void DrawLine(wxCoord x1, wxCoord y1, wxCoord x2, wxCoord y2,                 bool b_hiqual = true);
```

So it seems:
1. For the major grid line, double values are calculated, e.g. `334.89`, then there is an implicit cast to integer, which makes it value `334`.
2. For the minor grid line, the same value `334.89` is rounded to the nearest integer, e.g. `335`. Potentially that could explain the pixel difference.

sebastien-rosset, 2024-12-07:
@RooieDirk , if you can, would you be able to test the changes in #4249?

closed 2026-01-03 by RooieDirk