OpenCPN/OpenCPN #4232: OpenCPN 5.10.2 takes > 40 seconds to load on first run after reboot under X11

jimseng, 2024-11-28:
With a fresh bookworm install on a Pi 5 and fresh flatpak install of OpenCPN, when running it for the first time after a reboot it takes greater than 40 seconds to load. Running from the command line there is no output to the screen until the first window appears. After reboot and first run further launches of OpenCPN (flatpak) it loads normally (<6 econds) This happens when X11 is selected in raspi-config. Under Wayland it loads as expected.

bdbcat, 2024-11-28:
This is the loading time latency of the flatpak sandbox environment.
Not much that OpenCPN can do about that.
I agree that the X11 and Wayland difference is unexpected.  I would suggest to reconfirm that.

jimseng, 2024-11-29:
Hi.
This doesn't happen with 5.8.2. I tested a completely fresh Bookworm install and fresh flatpak install according to OpenCPN installation instructions. 5.10.2 > 40 seconds
Then I rolled back to 5.8.2 with:
flatpak update --user --commit=805191ab140141363a54f4be3d7fc47bf62e667d7f6159035b9d561a2dab94fd org.opencpn.OpenCPN 
Under Wayland and X11 with 5.8.2 the loading time after reboot is < 6 seconds.
I tried it several times to make sure I wasn't being an idiot.:-)

leamas, 2024-11-30:
hm..  As @bdbcat this is at least partly probably about your local environment. There are more users on RPi5 out there without these symptoms.

To debug, I'd suggest that you start opencpn from the command line like this:
```
$ rm -f ~/.var/app/org.opencpn.OpenCPN/config/opencpn/opencpn.log
$ date +%H:%M:%S; flatpak run org.opencpn.OpenCPN --loglevel debug
```
When opencpn startup is complete  with this long delay, just exit.  And please submit:
1) The time difference between the `date` command in the startup line and the first log entry in _~/.var/app/org.opencpn.OpenCPN/config/opencpn/opencpn.log_
2) The _~/.var/app/org.opencpn.OpenCPN/config/opencpn/opencpn.log_ logfile (github requires that it is compressed).

This should reveal where the startup delay is: in loading the system libraries or  something in opencpn's startup code.

jimseng, 2024-11-30:
date: 17:17:00
[opencpn.zip](https://github.com/user-attachments/files/17966225/opencpn.zip)

And as I mentioned. This doesn''t happen if I run 5.8.2

leamas, 2024-11-30:
hm... The first log entries are marked 17:11:53, but your start time is 17:17:00 which is about six seconds **after** the log?

"Confused"

leamas, 2024-11-30:
Otherwise, looking at the log there is:'
```
17:11:53.872 MESSAGE OCPNPlatform.cpp:726 Starting OpenGL test utility: "/app/bin/opencpn-glutil" opengl-info "/home/pi/.var/app/org.opencpn.OpenCPN/config/opencpn/gl_caps.json"
17:12:19.309 MESSAGE OCPNPlatform.cpp:923 OpenGL test complete.
17:12:19.309 MESSAGE OCPNPlatform.cpp:945 OpenGL determined CAPABLE.
```

This means that probing for OpenGL takes 26 seconds, that's the lion share of the total start  up  time of above 30 seconds.  And indeed, this check is something new in  5.10

@nohal: thoughts?

leamas, 2024-11-30:
@jimseng :

Please post the output from `time opencpn-glutil opengl-info`

jimseng, 2024-11-30:
Sorry. I tired this several times, I must have uploaded the wrong log. Here is the output from date +%H:%M:%S; 18:13:04
And just in case here is another log. There seems to be 26 seconds before the first line in the log.  And indeed, the log takes that long to appear in the folder. The second time I run it without a reboot the first log entry is exactly the same as the date +%H:%M:%S;
[opencpn.zip](https://github.com/user-attachments/files/17966361/opencpn.zip)
[opencpn-glutil.txt](https://github.com/user-attachments/files/17966382/opencpn-glutil.txt)

And if I run time opencpn-glutil opengl-info for a second time without a reboot:

real	0m0.144s
user	0m0.083s
sys	0m0.020s

leamas, 2024-11-30:
We have done what we can here for now, I guess. We need to get @nohal  a k a Pavel in the loop, he is the who wrote opencpn-glutils.

jimseng, 2024-11-30:
ok. Thanks for your input. I have a while before I start to rebuild my nav station so I am sat at home testing.

bdbcat, 2024-11-30:
You may, of course, run the opencpn-glutil from the command line, and time it's execution.  Might be instructive...

jimseng, 2024-11-30:
What's interesting is that the total time to launch OpenCPN after a reboot is about 45-55 seconds. If I run opencpn-glutil opengl-info first, then the launch time is more like 20 seconds, which is more or less the time it takes for the log file to appear. It's just a hunch and I am not familiar with flatpak at all but I wonder if the OpenGL thing is a red herring and it has something to do with running flatpak OpenCPN more generally. Why would opencpn-glutil opengl-info and running OpenCPN  be "instant-ish" the second time around? If it was the  loading time latency of the flatpak sandbox environment wouldn't I see a delay when running flatpak run --command=bash org.opencpn.OpenCPN? (I only get the prompt if I change --command=sh by the way). The prompt comes back pretty instantly.
And why not with OpenCPN 5.8.2? 
Anyway. That is probably enough for me tonight.

jimseng, 2024-12-01:
Just in case it is relevant. I have attached the output from journalctl. The mention of the word timeout reached makes my ears prick up:
Failed to create screen cast proxy: Error calling StartServiceByName for org.freedesktop.impl.portal.desktop.wlr: Timeout was reached
And .wlr... isn't that to do with wayland? I am running under X11 due to the tou
[journal.zip](https://github.com/user-attachments/files/17968812/journal.zip)
ch screen issue with wayland. Might not be of interest but I don't get much more than:
Dec 01 11:45:39 piplotter dbus-daemon[884]: [session uid=1000 pid=884] Failed to activate service 'org.freedesktop.impl.portal.desktop.wlr': timed out (service_start_timeout=120000ms) when I run it for the second time

jimseng, 2024-12-01:
This seems to have solved it:
systemctl --user mask xdg-desktop-portal-wlr
Takes about 4 seconds for opencpn to appear after reboot now.

leamas, 2025-04-22:
Thanks for input. Your solution is now documented in https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:program:flatpak_qanda#qis_there_a_way_to_handle_very_long_startup_times

Are you OK with closing this bug?

jimseng, 2025-04-23:
Yes. Thanks.

leamas, 2025-04-23:
OK. Closing. Thanks again!

closed 2025-04-23 by leamas