OpenCPN/OpenCPN #4236: [macOS] Window close button (red traffic light) non-functional

sebastien-rosset, 2024-12-02:
**Describe the bug**
The red close button (traffic light control) in the window title bar is non-functional across all OpenCPN windows on macOS, including the main application window and dialog boxes.

**To Reproduce**
1. Launch OpenCPN on macOS
2. Click the red close button (leftmost traffic light control) on any window
3. Observe that the window remains open

**Expected behavior**
Clicking the red close button should close the active window.

**Desktop:**
- OS: macOS (15.1)
- Device: MacBook Pro
- OpenCPN version: 5.11

**Additional context**
This appears to be a system-wide issue affecting all OpenCPN windows rather than being limited to specific dialogs or contexts.

bcn2, 2024-12-02:
Can't reproduce here, closes on red. 
macOS 15.1.1, mac Mini M1
OCPN 5.11.x

sebastien-rosset, 2024-12-03:
It looks like this may have been

sebastien-rosset, 2025-03-29:
I think this was a conflicting version of wxwidget on my system.

closed 2025-03-29 by sebastien-rosset