OpenCPN/OpenCPN #4242: Plugin API's broken in v5.11

TwoCanPlugIn, 2024-12-04:
Needs confirmation from other developers as I'm not sure if this reflects bugs in my own local OpenCPN build or in my plugin.

OS. Windows.

Local build and install of OpenCPN Master branch, commit b07e0c3

Test code uses plugin API 1.19.

A number of existing plugin API's are broken (ie not invoked). They worked correctly in the same test plugin when the plugin used API 1.18

LateInit 
SetupToolboxPanel (although I think this may have been broken aeons ago)
SetActiveLegInfo
SetPluginMessage.  

New plugin API's
GetSignalkPayload (not really broken, but ocpn_plugin.h lacks the linkage modifier extern DECL_EXP). Once corrected it works as expected.
GetPluginMsgPayload

I presume the usage of GetPluginMsgPayload is

`wxDEFINE_EVENT(EVT_OCPN_MSG, ObservedEvt);`
`PluginMsgId msg_id = PluginMsgId("WMM_VARIATION_BOAT");`
`listener_msg = std::move(GetListener(msg_id, EVT_OCPN_MSG, this));`
`Bind(EVT_OCPN_MSG, [&](ObservedEvt ev) {HandleMsgData(ev);});`
......

`void RacingPlugin::HandleMsgData(ObservedEvt ev) {`
`	PluginMsgId msg_id = PluginMsgId("WMM_VARIATION_BOAT");`
`	std::string message = GetPluginMsgPayload(msg_id, ev);`
`	wxLogMessage("Debug, GetPluginMsgPayload: %s", message);`
`}`

leamas, 2024-12-04:
> GetSignalkPayload (not really broken, but ocpn_plugin.h lacks the linkage modifier extern DECL_EXP).

hm... 
```
$ git blame include/ocpn_plugin.h | grep GetPluginMsgPayload
55b5e62968 (Alec Leamas                 2024-02-13 12:59:11 +0100 1871) extern DECL_EXP std::string GetPluginMsgPayload(PluginMsgId id, ObservedEvt ev);
```

That is, the ` extern DECL_EXP` is there and has been so since 2024-02-13.  Are you using a stale version of ocpn_plugin.h?

leamas, 2024-12-04:
> GetPluginMsgPayload

What is the actual problem here? Is there no debug message, or no/wrong data printed?

At a glance the usage looks ok to me. Not sure about the use of std::move but it should at least  not hurt, I guess.

leamas, 2024-12-04:
> That is, the  `extern DECL_EXP?  is there and has been so

.. but it is indeed missing in GetSignalkPayload. Sorry. Fixed in 7514a69ad

TwoCanPlugIn, 2024-12-04:
> What is the actual problem here? Is there no debug message, or no/wrong data printed

For GetPluginMsgPayload, the event is not fired and the handler is not invoked.

Possibly related, even though WANTS_PLUGIN_MESSAGING is set, SetPluginMessage is never invoked, so perhaps all OCPN Messaging is broken?

leamas, 2024-12-04:
AFK for a while, back later.

leamas, 2024-12-04:
> Possibly related, even though WANTS_PLUGIN_MESSAGING is set, SetPluginMessage is never invoked, so perhaps all OCPN Messaging is broken?

Too many turning wheels here....

> They worked correctly in the same test plugin when the plugin used API 1.18

Is that to say that it works in 5.10.2?  If so, could you make a bisect to find out when where this was broken?

TwoCanPlugIn, 2024-12-04:
I think it's a tad more convoluted as it's all run against v5.11.

Other than the inclusion of the new API 1.9 Specific API's, (PreShutdownHook, GetPluginMsgPayload & GetSignal?kPayload) there are only 4 code changes in my plugin, changing from API 1.18 to 1.19:
In CMakeLists.txt
`set(OCPN_API_VERSION_MINOR "18")`
`....`
`add_subdirectory(opencpn-libs/api-18)`
plugin header file
`class RacingPlugin : public opencpn_plugin_118, public wxEvtHandler {`
and plugin source file
`RacingPlugin::RacingPlugin(void *ppimgr) : opencpn_plugin_118(ppimgr), wxEvtHandler() {`

It makes little sense to me that a simple change in the OCPN API Version would result in my observations, which is why I suggested that another set of eyes confirms the bugs. I can't be sure that the bugs aren't due to a bug in my local build of OpenCPN nor in my plugin.

leamas, 2024-12-04:
OK, I'll try to find some time later. However,  no time today and probably not tomorrow either. We'll see...

TwoCanPlugIn, 2024-12-04:
It's not time critical, more of an FYI for you guys.

The test code is the RacingPlugin. I decided to reprise this plugin, which is of dubious value, but as an excuse to write a better example plugin, than the built-in demo plugin.

bdbcat, 2024-12-05:
Please check to be sure you have the latest commits from master branch.
Especially e5787ed21f5b8ea87d0afbc5893f3d76661a55b5

I can see a path where SetPluginMessage() would not be called on an API 119 plugin without this commit.

leamas, 2024-12-05:
> Especially e5787ed21f5b8ea87d0afbc5893f3d76661a55b5

There is no such commit. Perhaps you meant  be72dde2c and/or bd9ec10ca ?

bdbcat, 2024-12-05:
Ooops  
Pushed wrong branch here.
Fixed.
https://github.com/OpenCPN/OpenCPN/commit/e5787ed21f5b8ea87d0afbc5893f3d76661a55b5

TwoCanPlugIn, 2024-12-05:
With the latest build.

Fixed:
LateInit
SetActiveLegInfo
SetPluginMessage.

Not Fixed:
SetupToolboxPanel
GetPluginMsgPayload

Newly Broken:
AddOptionsPage

leamas, 2024-12-05:
I intend to make a stab at GetPluginMsgPayload problem, but not only that.

bdbcat, 2024-12-05:
I'll take a look at the others tonight.

TwoCanPlugIn, 2024-12-05:
No major rush, I doubt my test plugin will ever see the light of day.

leamas, 2024-12-05:
> GetPluginMsgPayload problem

This basically about bad documentation. The purpose of this API is to claim  messages  received over the htpp(s) REST interface. The testcase is in _test/rest_tests.cpp_ around line 215.

One obvious issue is to fix the comments in _ocpn_plugin.h_.

The other issue is about having this to work as you intended. It would be real nice, I guess. Practically, it's about adding a few lines to ` PlugInManager::SendMessageToAllPlugins` which drops the message in the message bus. However, I'm a bit worried about possible loops here, I don't have the complete picture.

@bdbcat : thoughts?

TwoCanPlugIn, 2024-12-05:
If GetPluginMsgPayload is meant to retrieve REST messages and not OCPN Messaging (OpenCPN Config, WMM_VARIATION_BOAT, OCPN_WPT_ACTIVATED etc), then I've definitely misunderstood the intent of this API.

Handling REST messages is not a feature that I am currently interested in, and just updating the comments in ocpn_plugin.h would clear up any confusion for other plugin developers.

leamas, 2024-12-06:
>  [...] and just updating the comments in ocpn_plugin.h would clear up any confusion for other plugin developers.

Fixed in 06fcf14a1

leamas, 2024-12-06:
See also #4245 which basically is about  making this work as you expected.

rgleason, 2025-04-22:
Not this one https://github.com/OpenCPN/OpenCPN/issues/4489

leamas, 2025-04-22:
@rgleason : Could you please, in detail, explain how that one has anything to do with this?

rgleason, 2025-04-23:
Its the api

leamas, 2025-04-23:
@TwoCanPlugIn : Ok to close?
@rgleason : Again: could you please inform us how this affects this bug in any way whatsoever?

TwoCanPlugIn, 2025-04-24:
Just compiled and installed the latest build.

Appears as though SetupToolboxPanel is still broken.

And because in my testplugin, it is where I invoke AddOptionsPage, I can't determine whether that too is broken.

As I'm the only person reporting this, and presumably because other "proper" plugins install their own tab on the Settings Dialog rather than add a panel to the User Options, perhaps its usefulness is obsoleted.

I do worry however about the large number of regressions introduced and also that it appears that many fixes and features are thrown over the fence without any unit tests to verify they actually work as intended.

Also looking at the new plugin API's. there is a lack of consistency.

Eg. 
```
extern DECL_EXP bool GetEnableTenHertzUpdate();
extern DECL_EXP void EnableTenHertzUpdate(bool enable);

```
What happened to good old Set and Get, as used appropriately in 
```
extern DECL_EXP bool GetEnableMainToolbar();
extern DECL_EXP void SetEnableMainToolbar(bool enable);

```

Although the use of two verbs in an API in itself is a bit mind boggling.

Perhaps they should be:
```
extern DECL_EXP bool GetMainToolbarVisibility();
extern DECL_EXP SetMainToolbarVisibilty(bool enable);
``` 
just like they are "properly" done in
```
extern DECL_EXP void SetTrackingMode(bool enable);
extern DECL_EXP bool GetTrackingMode();

```

bdbcat, 2025-04-25:
TwoCan

A bit confused by what you want to do.
At the moment, in O5.11, GetToolboxPanel() is never called by OCPN core.  I'm not exactly sure if it ever has been.
Have you used this API call before, in a previous version of O?

TwoCanPlugIn, 2025-04-25:
I've no idea if it ever worked. 

It's not so much a question of what I want to do, but the presumption is that if OpenCPN declares a virtual function that a plugin may override, it is assumed that it is there for a purpose and meant to do something.

If one API is no longer implemented, it then leads a plugin developer to think what other public API's are declared but no longer implemented. And if a plugin developer can longer trust that public API's perform as expected then that is a parlous state of affairs.

bdbcat, 2025-04-26:
I acknowledge that the available documentation for plugin API has been sparse over the tears, but is getting better.
But my question is this:  Using the available documentation, what is it that you expected SetupToolboxPanel() to do?

Reason I ask:  We either have to drop this function, or implement it.  Those are the choices.  Since you seems to be the only person ever to consider using this API, I thought you might have some insight.

TwoCanPlugIn, 2025-04-26:
I'm only "using" these API's in the Race Start plugin which I resurrected simply as a " better demo" plugin, using more of the API's than your built-in demo plugin. I also feel that Jon Gough's test_plugin is far too complicated and obfuscated to be of any use to a novice plugin developer.

I may update the repository, but I don't expect to release it via the plugin catalog, so I'm personally not fussed. For me it has just been a tool to experiment/test/use the different plugin API's.

With respect to some specific API's

GetToolbarToolCount
GetToolboxPanelCount;

If core doesn't invoke these overridden functions implemented by a plugin, or if it has no importance, then I guess they can be dispensed with. However I assume at some stage you must have implemented them for a reason.

GetToolbarItemId
Similarly, plugins return the id of the toolbar they install. What happens if a plugin installs multiple toolbar buttons ? Is this API of any use to OpenCPN?

OnSetupOptions 
AddOptionsPage
SetupToolboxPanel

If the expectation is that plugins install their own settings dialog, using ShowPreferencesDialog, rather than add a notebook tab to the options dialog, then these can be dispensed with, especially if no other plugins use them.

OnCloseToolboxPanel;

Similarly if you dispense with OnSetupOptions and SetupToolboxPanel then OnCloseToolboxPanel can also be removed (BTW, I don't know why OpenCPN chose to use its own return values rather than use the wxWidgets wxID_OK, wxID_CANCEL and wxID_APPLY constants)

SetDefaults

I have no idea whether any other plugin's use this, and it is probably superfluous as defaults can be configured when reading settings using wxConfig entries.

I must sound like a broken record, but API's last forever.

BTW, Sebastien has done an excellent job of documenting the plugin API's

bdbcat, 2025-04-26:
TwoCan...
These API do nothing now, and never have, AFAICT.  Must have been some very early idea, overtaken by events.
Clearly no-one is using them today, since the overridden methods in some plugin are never called by core.
GetToolboxPanelCount
SetupToolboxPanel
OnCloseToolboxPanel

These two APIs are in wide use:
OnSetupOptions
AddOptionsPage
For example, this is how chartdownloader_pi adds its control page to the main options dialog notebook.
Same for o-charts_pi, s63_pi.  Indeed, this is the only fully implemented method to add configuration pages for plugins.

For later reference, here is how that is implemented in a typical plugin:

```
void chartdldr_pi::OnSetupOptions(void) {
  m_pOptionsPage =
      AddOptionsPage(PI_OPTIONS_PARENT_CHARTS, _("Chart Downloader"));
  if (!m_pOptionsPage) {
    wxLogMessage(
        _T("Error: chartdldr_pi::OnSetupOptions AddOptionsPage failed!"));
    return;
  }

```
After this discussion, I expect I will remove the unused APIs from API-20, presently coming on O5.12.  No-one will miss them...

Thanks for your thoughts.

TwoCanPlugIn, 2025-04-26:
A couple of observations,
Sebastien has added the descriptive text for OnSetupOptions
"Called when OpenCPN preferences dialog is opened. Plugin can add its own
option panels to the preferences dialog. seems to be called once upon OpenCPN startup"

However in practice it seems that it is only called once upon System Startup. The "name" of the API could be considered to be misleading. It was implemented in 5.10 and seems to be implemented in recent builds of 5.11

If it is only called once, then why would a plugin choose this function in which to add an option page, rather than within the plugin Init or LateInit functions?

OnCloseToolboxPanel is currently implemented in 5.10. Surely it should be kept as how would a plugin that adds an Option Page  know when it has been closed with an OK or Cancel button and know whether to persist any changed settings?

To whit, The chart downloader panel. Add a catalog, then hit Cancel. 
Open the settings dialog again, open the chart downloader panel and the catalog that was previously added is still persisted, even though the dialog had previously been closed with the Cancel button. Cancel should mean discard any changed settings and do not persist them.

bdbcat, 2025-04-26:
""Called when OpenCPN preferences dialog is opened. Plugin can add its own
option panels to the preferences dialog. seems to be called once upon OpenCPN startup""

Not exactly correct.
OnSetupOptions() is called separately for each plugin _as it is loaded_.  This is THE time for each plugin to add it's page to the general Settings dialog.

TwoCanPlugIn, 2025-04-26:
Thanks for clarifying OnSetupOptions() and that it is the appropriate place for a plugin to call AddOptionsPage.

Conversely, when should the plugin call DeleteOptionsPage, DeInit, the new PreShutdownHook, or the Destructor?

bdbcat, 2025-04-27:
DeleteOptionsPage():  Should be called in plugin DeInit() method.  Imagine a plugin is disabled by user GUI, but OCPN core is not exiting.  We need for any added options page to be removed, since they are no longer valid.

DeInit()  Will be called by core when the plugin is disabled, generally preparatory to being unloaded/deleted.  Good place to persist plugin configuration parameters.

PreShutdownHook()  Called by core on enabled plugins just before OCPN exits.  Allows plugin to do misc. internal cleanup actions, as necessary.  This happens before DeInit().  Rarely needed.

Plugin DTOR:  Called by core as last action while unloading the plugin.  At this point, all that might be necessary is to free any stray memory allocations to avoid leaks.

leamas, 2025-04-27:
Good info, some of which should be used to update the API docs.

That said, yet another missing piece of documentation is "Plugin lifecycle" with an overview how plugins are loaded, enabled, disabled and unloaded.

leamas, 2025-04-27:
Filed #4504 in an attempt to make some actual updates related to this discussion. 

Will not merge unless someone has time to review it. More than open to more actual ideas how to update these docs.

rgleason, 2025-04-27:
Rather than restart after a crash, useable copy of the navobj.changes file is needed
https://github.com/OpenCPN/OpenCPN/issues/4345
See Pavel's comment here https://github.com/OpenCPN/OpenCPN/issues/4345#issuecomment-2619353594

leamas, 2025-05-03:
@TwoCanPlugIn : With new comments in place: should we close, or do you see things here not being handled?

TwoCanPlugIn, 2025-05-04:
Closing as suggested.

I don't actually use these API's in production for either the Engine Dashboard or TwoCan. I only included them in my Race Start plugin which I simply resurrected as a platform to discover what many of the plugin API's actually did. I also thought that the Race Start display may be a "better" demo plugin for future plugin developers.  Now that the Plugin API's have been documented it is kind of moot.

I'm in the process of getting my boat ready to launch for the season, so can't devote much more time to this. As indicated if others find API's are broken, they can file specific issues.

closed 2025-05-04 by TwoCanPlugIn

rgleason, 2025-05-04:
Thanks Steve for your persistence.