OpenCPN/OpenCPN #4257: When I tried to compile Android, it failed to compile

Twillpower, 2024-12-11:
When I compile Android, the compilation fails. I use release 5.10.2, and the error can not find chart1.h. I found it has been deleted in the previous commit, do I need to modify ocpn_wrapper.cpp, how to modify

leamas, 2024-12-11:
Basically a duplicate of #2081. I suggest that you close the  issue here and continues the discussion in the existing one.

stone1706, 2024-12-11:
Me too,  i got same Error...

Twillpower, 2024-12-11:
I browsed through the issue you provided, but did not find the part related to my problem, and did not find the relevant solution. My error was caused by the step make -f Makefile.android, when it compiled ocpn_wrapper.cpp, it could not find chart1.h. I found that chart1.h was removed two years ago but ocpn_wrapper.cpp was last modified 9 years ago

leamas, 2024-12-11:
This is a general problem, and I doubt there is a quick fix available. Instead of filing new issues add your problem to #2081. This is all about the Android build instructions which basically is a mess, known issue.