OpenCPN/OpenCPN #4258: Build Android APK failed.

stone1706, 2024-12-11:
**Describe the bug**
Hi ,
I tried to follow the README  (https://github.com/OpenCPN/OpenCPN/tree/master/buildandroid) to compile Android apk,but it failed to compile.
 Following the readme file , step by step ,when I got this step "build APK"
********************************************************************************************
build apk
 a)  Use qmake to build a makefile (called makefile.android)
            This makefile will combine the OCPN static libraries, the wxQT libraries,
            and the Qt libraries into a single dynamic (libopencpn.so) library.
            Also included is a "wrapper" object which effectively provides the
            entire application with a main() entry point, and calls the opencpn OnInit() method.
            qmake uses a project file called opencpn.pro for its instructions.
            We will need to edit that file in order to set some paths and parameters. 
........
********************************************************************************************
I got error message, told me : not found chart1.h ???
I checked out  buildandroid/ocpn_wrapper.cpp file 
the code line 34 : #include "chart1.h"
what I know chart1.h has been removed from project about 2 years ago, right ?  ....
why  build APK still need this old file and get error .

I notice there is new version APK in GooglePlay ,it build in 12/2024, so  there should be a new buildandroid readme file updated  or  someone  can help me to modify the project to build APK? 

Best regards

leamas, 2024-12-11:
There is now three bugs covering the same thing. There is no way we can have a productive discussion this way. Please close this bug and continue the discussion in the original #2081 issue.

stone1706, 2024-12-12:
where is the bugfix

closed 2024-12-12 by stone1706