OpenCPN/OpenCPN #4259: Cannot Update Plugin

CY-Dash, 2024-12-12:
version 5.10.2
When I tried to update the plugin, it displayed as shown in the picture.
Also, when I click on the settings, I cannot enter.

[image]
[image]

leamas, 2024-12-12:
I cannot reproduce this. Questions:
1. Do you still get this error?
2. Are you behind some kind of firewall?

transmitterdan, 2024-12-12:
What happens when you click this URL: https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml

CY-Dash, 2024-12-13:
1:Yes, ever since I downloaded the software, it has been like this for several days without success.
2.I tried to turn off the firewall on my computer and use VPN, still couldn't update it.
3.Is there any other way to manually install the plugin? I have already downloaded the plugin installation package.

CY-Dash, 2024-12-13:
> What happens when you click this URL: https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml

I can open the link normally, with the firewall turned on and without using VPN.

bdbcat, 2024-12-13:
Manual workaround.
1.  Stop opencpn
2. Download the file at link https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml
3. Copy the file into the same directory that contains your config file, overwriting the existing ocpn-plugins.xml
4. Start opencpn as normal.  The new ocpn-plugins.xml will be assimilated, and hopefully you will be able to install plugins.

Just so you know, this bug is almost certainly a result of your particular VPN and/or firewall settings.

CY-Dash, 2024-12-13:
> Manual workaround.
> 
> 1. Stop opencpn
> 2. Download the file at link https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml
> 3. Copy the file into the same directory that contains your config file, overwriting the existing ocpn-plugins.xml
> 4. Start opencpn as normal.  The new ocpn-plugins.xml will be assimilated, and hopefully you will be able to install plugins.
> 
> Just so you know, this behavior is almost certainly a result of your particular VPN and/or firewall settings.

Thank you very much, the other plugins have finally been displayed and can be downloaded normally. But when I click the update plugin button, it still pops up and cannot download from the url.

closed 2024-12-13 by CY-Dash