OpenCPN/OpenCPN #4271: BUG: Options >Plugins  Updates available in catalog, do not show in the PI List

rgleason, 2024-12-19:
**Describe the bug**

Please refer to this report from Corsair  Post #62

https://www.cruisersforum.com/forums/f134/ais-radar-plugin-missing-information-190125.html#post3958216

I do not believe this is a plugin bug.

rgleason, 2024-12-19:
Not a bug.  The green update arrow works with other plugins, but not AISradar.
Filing this with AISradar.

Hakansv, 2024-12-21:
For info was the same seen today on Win for radar_pi update to version 5.5.5. This was reported on CF and I also saw it on a W10 today.
I suppose it's a change in the general plugin-managing rather then each plugin or OCPN itself?

@rgleason we may open this issue again?

rgleason, 2024-12-21:
Yes will do

Hakansv, 2024-12-21:
Thanks. Also discussed [here](https://opencpn.zulipchat.com/#narrow/channel/432618-OCPN-5.2E10-Release-plan/topic/Plugin.20update.20problems/near/490276884)

rgleason, 2024-12-29:
See this CF Post by Corsair https://www.cruisersforum.com/forums/f134/plugin-testing-all-operating-systems-needed-291446.html#post3960246

rgleason, 2025-03-23:
The issue with AISradar is fixed, also bdbcat and Mike Rossiter are aware of the problem with SD frontend and using v in tags. 
Leamas has fixed this with a clever workaround, but I think it should also be fixed in SD metadata output.

@bdbcat Can I close this now?

bdbcat, 2025-03-23:
Rick...
"Leamas has fixed this with a clever workaround, but I think it should also be fixed in SD metadata output."
No.  The current templates and workflow using "v" in tags and metadata is just fine.  Leamas fixed a plain bug in the code relating to this.
Anyway, this is ready to close now.

rgleason, 2025-03-23:
Ok, my misunderstanding. Closing.

closed 2025-03-23 by rgleason