OpenCPN/OpenCPN #4277: MacOS (15.1.1) clean build of OpenCPN 5.11.0-0 chart offset and inaccurate point selection

cpelon, 2024-12-24:
**Describe the bug**
I am having an issue with chart offset and inaccurate point selection. Also in settings ->user interface -> general options the scale factor sliders are not showing correctly and I can not enable OpenGL under display settings.

I am on M1 MacBook on Sequoia 15.1.1. Followed the build instruction at 
https://opencpn-manuals.github.io/main/opencpn-dev/mac-osx.html

Attached video

https://github.com/user-attachments/assets/a85f9f8c-1487-471a-809c-2f22582f60ff

sebastien-rosset, 2024-12-24:
Possibly related to #4219 and #4051

cpelon, 2024-12-24:
Thanks Sebastien! If I run OpenCPN from /tmp/opencpn/bin/OpenCPN.app it quits unexpectedly (due to security since the app is not trusted/signed). I can run from OpenCPN/build folder by double clicking on the icon, but if I try to run from terminal to add "-p" it quits unexpectedly. I see there are some scripts in the ci folder. I may need to run these??

cpelon, 2024-12-24:
I think running ci/universal-build-macos.sh may have resloved my issue.

bdbcat, 2025-03-22:
Closing, reported resolved

closed 2025-03-22 by bdbcat