OpenCPN/OpenCPN #4286: Bug/RFE  Plugin Manager -Update plugin catalog, upgrade plugin is not shown, have to uninstall

rgleason, 2024-12-31:
This may not be a Plugin Manager issue, but please see 

BUG/RFE Update Catalog for new version and must uninstall first. #61 
https://github.com/rgleason/vdr_pi/issues/61

Had vdr_pi 1.3.40 (imported) to start, and then update catalog, failed to pickup that there was an Upgrade available and we knew there was one, so we uninstalled and then when enabling the plugin again, we found the new version and selected that.

rgleason, 2025-01-01:
https://www.cruisersforum.com/forums/f134/plugin-testing-all-operating-systems-needed-291446.html#post3961042

Hakansv, 2025-01-21:
I've examples to emphasize the trouble:
On W10

Here the update available icon should have been shoved since there are a newer version available if I click "reinstall". To click the reinstall button at this state is not obvious.
[image]

Here all is fine. The update available icon is present. Why is it different from the above example?
[image]

rgleason, 2025-01-22:
Please see this. All dups have been removed. Green arrow issue is no better.
https://www.cruisersforum.com/forums/f134/plugins-updates-291981.html#post3966769

rgleason, 2025-01-22:
Is there a lookup list of valid installed plugins? Compare that against the master catalog version numbers,  if the version in the master catalog is greater then show a green upgrade arrow. 

Do nothing else with fancy buttons, just let then uprade or install a different version, by selection of which one 

The buttons might have change a little, but keep it simple.

Notify them but let them pick which version.

bdbcat, 2025-01-22:
" install a different version, by selection of which one"
No.  There will only ever be exactly zero or one fresh plugin in the master catalog.

This is simple UX doctrine.  Virtually no modern app installer requires the typical user to know _a priori_ which of multiple versions of the code that she wants/needs.

bdbcat, 2025-01-22:
"Why is it different from the above example?"
As Rick has discovered, probably some interaction with the tarball cache logic.  Clearly a bug.

rgleason, 2025-01-22:
Now after restarting OpenCPN twice, I see that Climatology is showing a green arrow. This behavior is not helpful to users!  It should work directly after the catalog is updated.  

[image]

Just one choice is given here. (before the second OpenCPN startup, it showed two) I think any other availble choices should be listed from the cache.
[image]

Something needs to be refreshed after the new catalog is loaded, and a new search through the Lookup Table of existing installed plugin versions and comparing that with the new Catalog versions. 

 Will put some sceenshots up. Also some of the other TP plugins show green arrows but some Plugins don't..  Another specific plugin, Tactics which still is not showing a green arrow, after uninstalling and reinstalling.

Also after updating Climatology (had a green arrow) I could not uninstall the update and go back to the previous version.  This is not progress I think. What if the user finds a updated plugin isn't working? The first thing they think of is going back to the previous one. Its perfectly logical and should be allowed!!!!

Tactics isn't showing a green arrow after trying everything. Enable, disable, uninstall, etc. Its being a real bad boy.

[image]

[image]

rgleason, 2025-01-22:
I have noticed that once a user updates to a new version, they are no longer allowed to pick an earlier version even if it exists in the Cache under 

C:\ProgramData\opencpn\plugins\cache\tarballs.

We expect users to be testing the plugins. We have not extensively tested each and every plugin. Therefor when a user finds that a plugin does not work, he without fail wants to go back to the previous plugin.

The present interface does not permit or allow this easily because none of the tarballs in the cache are listed as part of the selection.

What do you expect the users to do in this case? Use Import Plugin Tarball and navigate or browse to 

C:\ProgramData\opencpn\plugins\cache\tarballs

This is not intuitive and will be a PAIN.

bdbcat, 2025-01-22:
I agree that the current implementation is not great, and has some holes.  But "will be" is not quite the right tense.  This is OCPN 5.10.2, in production since last July.  The bug somehow not noted in 5.10.2 Beta testing.
I'd certainly argue that 7 months of production without panic over this bug (until now) is meaningful, at least in terms of priority.

Anyway, it is a legitimate bug, logged in the correct place.
Now, as to the resources.....

rgleason, 2025-01-22:
Agreed, but there has been some rumbling from users, never definitive though.

bdbcat, 2025-03-17:
Rick...
Started studying this issue.  Presently cannot reproduce.  So reading code....

The first point is that the PIM code is really expecting NOT to see multiple versions of the same plugin in the catalog.  If it sees that, it could (and does) somewhat randomly choose whether or not to show the green upgrade arrow in the PIM panel.  Depends solely upon the order in which the compatible plugins appear in the catalog.

So let us agree, first, that multiple identically named plugin metadata entries in the catalog will cause undefined operation of PIM.  Plugin publishers must follow this rule for predictable functionality.

There may be other logic corner cases contributing to user confusion here.  Will continue to investigate.

rgleason, 2025-03-17:
Totally agree about not having multiple versions of the same plugin.  

Shouldn't I close this now and wait to see if it comes up in some other way?

bdbcat, 2025-03-17:
Yes, please close for now.
**If the issue appears again, at any time, we will need a copy of the "ocpn-plugins.xml" taken at the moment of the error from the target system in order to fully debug and understand.**

Hakansv, 2025-03-17:
@bdbcat 
I may have an example. 
msvc-wx32:10.0.19045

No green arrow for the available update as of the shot.
It might though be I've use a tarball sometime before that can disturb?

[ocpn-plugins.xml.txt](https://github.com/user-attachments/files/19290021/ocpn-plugins.xml.txt)

[image]

leamas, 2025-03-17:
> It might though be I've use a tarball sometime before that can disturb?

If you have imported a tarball with a version >= 1.3.65.0 it will rightfully "disturb" so to speak. When imported, a new dynamic catalog entry based on the imported version is added to the catalog. So, if the imported tarball had version v1.3.65.1 the catalog entry v1.3.65.0 will not be deemed as an update.

That said, 1.3.62 is certainly less than 1.3.65  which should be visible as an update. But then again, the version 1.3.62+492.v1.3.62.0 looks utterly broken in my eyes. Shouldn't it be 1.3.62+492?

This is is all about comparing versions...

rgleason, 2025-03-17:
Sorry, I am not following this very well. "Its all about version numbers.. "
Should I  close?

closed 2025-03-17 by rgleason