OpenCPN/OpenCPN #4291: Chart downloader stops downloading Full OSM base shape map

mgrouch, 2025-01-04:
Chart downloader stops downloading Full OSM base shape map

On pi5 Bookworm arm64. OpenCPN from .deb package (backports)

Thanks

nohal, 2025-01-05:
Not reproducible, downloaded just fine on exactly the same system connected via Starlink.
Just as a note, the full resolution OSM dataset requires very powerful HW to be useful, a RPi is not such a system and trying to use it there is IMHO pretty much a waste of effort.

closed 2025-01-05 by nohal

mgrouch, 2025-01-05:
Pi5 is very powerful. Which system did you use to reproduce?

thanks

nohal, 2025-01-05:
If I say exactly the same I mean exactly the same, RPi 5 with Bookworm and the package you used.
Your perceprion of "very powerful" can certainly be corrected by looking at pretty much any benchmark of a RPI 5 and comparing it even with results of something as low-end as https://www.intel.com/content/www/us/en/products/sku/231803/intel-processor-n100-6m-cache-up-to-3-40-ghz/specifications.html

mgrouch, 2025-01-15:
I've tried aquamap application on even less powerful hardware than pi5. It handles full resolution OSM just fine.
It even allows titling maps in 'highway' kind of mode with adding a perspective.

nohal, 2025-01-15:
Are you sure Aquamap uses Shapefile data? I really do not understand why I even react this irrelevant comment to be honest.

rgleason, 2025-01-15:
Pavel, Aquamap is pretty good, its worth taking a look at it to see what they do.