OpenCPN/OpenCPN #4292: De-referencing erased object pointed to by interator is an error

transmitterdan, 2025-01-04:
https://github.com/OpenCPN/OpenCPN/blob/1b0c1ce672746d77bc5169648c7ae2f142f0c009/gui/src/connections_dialog.cpp#L526

I believe that once the pointed to object is erased via iterator 'found' then further references to 'found' is an error.  The next line would fail I think.

https://github.com/OpenCPN/OpenCPN/blob/1b0c1ce672746d77bc5169648c7ae2f142f0c009/gui/src/connections_dialog.cpp#L527

leamas, 2025-01-04:
Good catch. Thanks, will fix during the day

closed 2025-01-04 by leamas

transmitterdan, 2025-01-04:
I suspect either it should fail (which is does in a VS2022 debug build) or maybe the dereference silently fails in a 'release' build. Also, maybe the expected function call never happens.

transmitterdan, 2025-01-04:
While you are looking at that, I think this is another nearby error:
https://github.com/OpenCPN/OpenCPN/blob/1b0c1ce672746d77bc5169648c7ae2f142f0c009/gui/src/connections_dialog.cpp#L563

It seems that index is -1 which is not a valid array index. I don't really follow what this is supposed to be doing. Maybe forgot to set index somewhere after it was initialized to -1?

leamas, 2025-01-05:
Looks like a mess... I'm currently working on #4283 which will make this code go away completely.

But until then we need to fix this... What do you think of this patch?

```
diff --git a/gui/src/connections_dialog.cpp b/gui/src/connections_dialog.cpp
index 124fec019..29b5934e6 100644
--- a/gui/src/connections_dialog.cpp
+++ b/gui/src/connections_dialog.cpp
@@ -545,7 +545,7 @@ void ConnectionsDialog::OnEditDatasourceClick(wxCommandEvent& event) {
     int index = -1;
     auto found = std::find(TheConnectionParams().begin(),
                            TheConnectionParams().end(), mSelectedConnection);
-    if (found != TheConnectionParams().begin() && (*found)) {
+    if (found != TheConnectionParams().end() && (*found)) {
       ConnectionEditDialog dialog(m_parent);
       dialog.SetSize(
           wxSize(m_parent->GetSize().x, m_parent->GetSize().y * 8 / 10));
@@ -556,11 +556,12 @@ void ConnectionsDialog::OnEditDatasourceClick(wxCommandEvent& event) {
       auto rv = dialog.ShowModal();
       if (rv == wxID_OK) {
         ConnectionParams* cp_edited = dialog.GetParamsFromControls();
-        delete (*found)->m_optionsPanel;
         StopAndRemoveCommDriver((*found)->GetStrippedDSPort(),
                                 (*found)->GetCommProtocol());
-
-        TheConnectionParams()[index] = cp_edited;
+        delete (*found)->m_optionsPanel;
+        delete *found;
+        TheConnectionParams()[found - TheConnectionParams().begin()] =
+            cp_edited;
         cp_edited->b_IsSetup = false;  // Trigger new stream
         FillSourceList();
         UpdateDatastreams();

```

commit b1b62f27 (2025-01-04): connections_dialog: don't dereference erased iterator - #4292

Kudos: Dan Dickey

Closes: #4292