OpenCPN/OpenCPN #4297: Improve route dynamic waypoints: inconsistent 001/NM001.

nocodehummel, 2025-01-07:
Re-sequence of route way-points is limited to way-points that are not named specifically. This could be improved by checking the first part of the way-point name for the dynamic name. When the first part is a number it could be re-sequenced. 

Named way-points that start with a numeric sub-string should be included when the route is re-sequenced. The user can prevent renumbering of a specific way-point by changing the first part of the name. 

A minor issue is also that the initial name of way-points is inconsistent. When a new route is created the way-points are numbered with 3 characters including leading zeros. When a way-point is inserted afterwards the name starts with *NM* followed by a number. The route way-points are not re-numbered when a way-point is inserted. Way-points should be numbered consistent and automatically re-numbered.

Hakansv, 2025-01-07:
Try a route right click:

[image]

nocodehummel, 2025-01-07:
@Hakansv Yes, I know the feature, but it only works when the name is 123 or NM123. I think it should also work when the name is 123-Sandbank.

Hakansv, 2025-01-07:
The WP naming also has certain rules to follow when exporting to Garmin and the like. I don't know all the details but there are others here.

nocodehummel, 2025-01-16:
Resolved by merged #4301.

closed 2025-01-16 by nocodehummel