OpenCPN/OpenCPN #4300: The Connections  | Edit dialog is broken

leamas, 2025-01-10:
**Describe the bug**
The Connections | Edit dialog is broken is several ways. This is includes that the values set by user  are not actually used after closing the dialog. Here are is also layout issues making it it impossible to create a dialog which has the standard layout normally used bu wxWidgets. Net result is that the dialog cannot be resized and probably also not adapts that well not small (Android) or high DPI screens.

**To Reproduce**
Steps to reproduce the behavior:
1. Options | Connections | Edit
3. Change arbitrary value
4. Note that dialog  cannpt be resixed
4. Click OK
5. Note that the changed value is not actually used.

**Expected behavior**
The dialog should be a "nomal" wxWidgets dialog which has a layout based on the contained widgets.  It should be possible to resize. The values changed should be used.

**Additional context**
Discussed in https://github.com/OpenCPN/OpenCPN/pull/4283#issuecomment-2581172852
The filter setup is basically outdated. While the new communication interfaces supports a much more than N0183 this is not reflected in the filter setup.

Hakansv, 2025-01-13:
As discussed here https://github.com/OpenCPN/OpenCPN/pull/4283#issuecomment-2586925946 I've tried to bisect when the connection edit bug was introduced:
The fault is merged to master between 2024-12-28 08:00 UTC and 2024-12-29 18:00 UTC
(When I run the latter merge, N0183 connection's edit don't open dialog. TCP edit opens but don't change on OK.)
so I suppose either cd353db29 or 2ca798bea or f88f9a22c8 is the cause?

There are for me to much and sophisticated changes there to find out the bug cause.
I missed to test to edit connections after it was merged. Possibly not tested enough by the author either?

bdbcat, 2025-01-13:
Thanks to Hakan for isolating the offending commit.
Corrected by https://github.com/OpenCPN/OpenCPN/commit/dd20c0830a6d1336165d7c0992a4c9afd1abf562

leamas, 2025-01-13:
... and  3ba600f. Closing

closed 2025-01-13 by leamas