OpenCPN/OpenCPN #4306: RFE  Screen scaling for Windows 125-250-300% etc

rgleason, 2025-01-13:
**Is your feature request related to a problem? Please describe.**
I would like to be able to enter my recommended Windows Screen Scaling factor (100%, 125%, 150%,200%, 250%, 300% etc)
under Options > Display, in order to have all text, icons, plugins etc respond accordingly.

Hakansv, 2025-01-13:
I think basically Windows and GTK3 are reporting this to OCPN and it's used there and in the plugin API. (OCPN_GetWinDIPScaleFactor / OCPN_GetDisplayContentScaleFactor) So it may be about actually using the existing wheel.

rgleason, 2025-01-13:
I hope so, Is there any documentation about this and some code packages that we can use in the plugins?
What about the other OS?  Mac and Linux?

Hakansv, 2025-01-13:
GTK3  is about some MacOS

rgleason, 2025-01-15:
Please refer to Discussion https://github.com/OpenCPN/OpenCPN/discussions/4312