OpenCPN/OpenCPN #4314: RFE Windows 11 fast Dell Toolbar Hamburger Menu takes ages to load.

rgleason, 2025-01-15:
I have noticed this for quite awhile, but decided it was not that important. I think it should be improved however, if possible.

bdbcat, 2025-01-16:
Not sure I understand.
Which hambuger?
Takes a long time to activate when touched, or long time to appear on startup?

rgleason, 2025-01-16:
Long time to appear on startup.

bdbcat, 2025-01-16:
Hamburger will not appear until all enabled plugins are loaded and stabilized.  As designed.
Maybe one plugin is very slow?

rgleason, 2025-01-16:
Long time to appear on startup. Will check.

transmitterdan, 2025-01-17:
It's probably the plugins.  Try it with no plugins installed and it pops right up.

bdbcat, 2025-01-17:
Agree.
But which plugin is the worst offender in startup time?

rgleason, 2025-01-17:
No plugins active.
Still slow loading Toolbars.
Have tried removing chart groups and chart directories

- Pilot Charts,
- Inland
- Inland buoys
- MBtiles with Noaa RNC as tiles

Now have just the following
GSHHG,  US_CG01, 05, 07, 08, 09, CMP

Still slow.

bdbcat, 2025-01-17:
How slow?
From double click on desktop icon until the Hamburger is ready, how many seconds?
For average W10 laptop, SSD drives, loading cm93 only, no plugins:  I get about 7-8 seconds.

bdbcat, 2025-01-17:
You may also examine the logfile.  It has timestamps that will help isolate the bottlenecks.

transmitterdan, 2025-01-17:
My W11 machine with SSD and Intel(R) Core(TM) Ultra 7 155U   2.10 GHz takes about 4 seconds to load.

It may also be an anti-virus that is spending time studying the .exe file and DLLs before it lets Windows actually load it.  I have other machines running Sophos and it really slows down startup of some programs as it scurries around examining DLLs and EXEs before it will let Windows run them.

rgleason, 2025-01-17:
Here are the other ini and log files. Before and after.

[opencpn-19.0-second-1-17-24.ini.zip](https://github.com/user-attachments/files/18454489/opencpn-19.0-second-1-17-24.ini.zip)
[opencpn 6.5-seconds.log.zip](https://github.com/user-attachments/files/18454490/opencpn.6.5-seconds.log.zip)
[opencpn-6.5-seconds.ini.zip](https://github.com/user-attachments/files/18454491/opencpn-6.5-seconds.ini.zip)

Now inspecting.

rgleason, 2025-01-17:
I think the critical part is in here:

```
07:30:01.839 MESSAGE OCPNPlatform.cpp:945 OpenGL determined CAPABLE.
07:30:03.078 MESSAGE options.cpp:4897 options: got device count: 1
07:30:03.649 MESSAGE ocpn_frame.cpp:4685 OnInitTimer...5
07:30:03.776 MESSAGE ocpn_frame.cpp:4685 OnInitTimer...6
07:30:03.887 MESSAGE ocpn_frame.cpp:4685 OnInitTimer...7
07:30:03.887 MESSAGE ocpn_frame.cpp:4979 OnInitTimer...Last Call
07:30:03.887 MESSAGE ocpn_frame.cpp:4997 OnInitTimer...Finalize Canvases
07:30:03.902 MESSAGE glChartCanvas.cpp:562 BuildFBO 3
07:30:03.902 MESSAGE glChartCanvas.cpp:1016 OpenGL-> Framebuffer OK, size = 3838
07:30:03.934 MESSAGE glChartCanvas.cpp:562 BuildFBO 3
**07:30:03.934 MESSAGE glChartCanvas.cpp:1016 OpenGL-> Framebuffer OK, size = 3838**
**07:30:10.099 MESSAGE plugin_loader.cpp:787 PluginLoader: Deactivating PlugIn: C:\Program Files (x86)\OpenCPN 5.10.2-0+12e5712\plugins\chartdldr_pi.dll**
07:30:10.099 MESSAGE chartdldr_pi.cpp:223 chartdldr_pi: DeInit
07:30:10.099 MESSAGE plugin_loader.cpp:787 PluginLoader: Deactivating PlugIn: C:\Program Files (x86)\OpenCPN 5.10.2-0+12e5712\plugins\wmm_pi.dll
07:30:10.099 MESSAGE ocpn_frame.cpp:1691 opencpn::MyFrame exiting cleanly.
07:30:10.186 MESSAGE glChartCanvas.cpp:562 BuildFBO 3
07:30:10.186 MESSAGE glChartCanvas.cpp:1016 OpenGL-> Framebuffer OK, size = 3838
07:30:10.267 MESSAGE glChartCanvas.cpp:562 BuildFBO 3
07:30:10.278 MESSAGE glChartCanvas.cpp:1016 OpenGL-> Framebuffer OK, size = 3838
07:30:10.351 MESSAGE chartdb.cpp:287 Chart cache PlugIn purge
07:30:10.351 MESSAGE ocpn_frame.cpp:1870 gFrame destroyed.
07:30:10.372 MESSAGE ocpn_app.cpp:1948 opencpn::MyApp starting exit.
07:30:10.374 MESSAGE ocpn_app.cpp:1988 LOGBOOK:  2025-01-17 12:30:10 UTC OFF: Lat   38.78970 Lon  -43.59770
07:30:10.374 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: C:\Program Files (x86)\OpenCPN 5.10.2-0+12e5712\tcdata\harmonics-dwf-20210110-free.tcd
07:30:10.380 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: C:\Program Files (x86)\OpenCPN 5.10.2-0+12e5712\tcdata\HARMONICS_NO_US.IDX
07:30:10.385 MESSAGE ocpn_app.cpp:2016 opencpn::MyApp exiting cleanly...

07:30:26.707 MESSAGE ocpn_app.cpp:1126 
```

rgleason, 2025-01-17:
Looking at the original log file[ here ](https://github.com/OpenCPN/OpenCPN/issues/4314#issuecomment-2598248458)  I don't see the 19 seconds for the Toolbar Menu to appear, which is a little perplexing.

After renaming the opencpn.ini and rebooting the menu loads much faster. I have also configured my original chart directories (using chart subdirectories and Chart Groups) and the appearance of the Toolbar is still very good.

Then I updated all plugins leaving just WMM and Chartdownloader on.  The toolbar is still appearing very fast.

I was unable to determine what the problem was.

rgleason, 2025-01-17:
See this VDR bug No File Loaded https://github.com/rgleason/vdr_pi/issues/82

@bdbcat Should I remove VDR 1.3.53 from the catalog?

rgleason, 2025-01-18:
VDR has been removed from the catalog.
I do not know if that was the cause of the 19 second load time for Toolbars.
Closing

closed 2025-01-18 by rgleason