OpenCPN/OpenCPN #4319: Celestial Navigation Plugin shows incorrect altitude (Hc) on FIND celestial body form

rbossert149419, 2025-01-17:
Celestial Navigation Plugin shows Altitude of the Body on the Find celestial body form.  Altitude is also known as Hc.  The value shown is wrong.  There are 3 examples that illustrate.  They are attached. 

**To Reproduce**
Steps to reproduce the behavior:
Drop a Mark to denote your DR.  Move your Boat to the Mark (as close as possible).  Then enter a sight, select Sun, Planet, or Star.  Enter your sight data.  Select the FIND (Find celestial body) button on the site tab.  The field, Altitude, is incorrect and inconsistent to the Line of Position OpenCPN Plots on the chart.  The Line of Position plot is correct.  The problem is the use of Law of Cosines formula by Open CPN in the FIND form.  When plotting the LOP, Open CPN had the correct Altitude and Azimuth.  II attach 3 examples that illustrate Open CPN's calculation of Altitude (Hc) and the correct answer.  In the 3 examples, the error resulted in a 3-6 NM error in the EP or Fix.  The Line of Position is correct.  Meanwhile, people have to ignore the FIND Celestial Body button and deduce Hc by using the measurement tool and the Plotted LOP.

**Expected behavior**
The Altitude should be within .1 minute of a degree compared to Commercial Addition of Nautical Almanac using Sight Reduction formula pages 277-283.  To calculate Hc using Law of Cosines, you use LHA, Declination, and Latitude.  The calculation should use 5 decimal accuracy for LHA, DEC, and LAT to get to .1 minute precision.

**Screenshots**

[Open CPN Altitude calculation error PDF.pdf](https://github.com/user-attachments/files/18447270/Open.CPN.Altitude.calculation.error.PDF.pdf)

**Desktop (please complete the following information if applicable):**
 - OS: macOS, Windows 10.  I think it is true on all.
 - Version 2.4.41

rgleason, 2025-01-17:
This should be filed with the plugin repository Issues for Celestial Navigation . Not in the main OpenCPN Issues.  Thank you.
Please close this after you transfer it to Cel_nav.

rbossert149419, 2025-01-17:
Do you know the folder name.  Is the path open on/cel_nav  ?

leamas, 2025-01-17:
tl; dr;  https://github.com/rgleason/celestial_navigation_pi 

The hairy stuff: it's about a repo, not folder. The canonical way is to look into the _ocpn-plugins.xml_ file  which is available in the config directory, _~/.opencpn_  for regular linux. grep for "celestial" and you will find multiple entries, one for each platform. Look for  `<source>`  in any of them.

rbossert149419, 2025-01-17:
Thanks for the response.  I just loaded the bug to the URL you supplied
me.  Let me know that where I placed the issue is satisfactory.   And then,
I'll close the bug I opened in the OpenCPN issues list.

Bob Bossert

rgleason, 2025-01-17:
Thank you rbosset.  Looks like it is in the right place.  Just to let you know, we currently have no one able to fix this.

rgleason, 2025-01-17:
I will close this for rbosset

rgleason, 2025-01-17:
Yes, indeed, please submit them. We need experts such as you!  I hope that some old programming sailors who are passionate about sextants and accuracy will focus on improving this tool, as it could become a great resource. 

Also if you have any suggestions or improvements to the manual, please advise.

How to get to the manuals from OpenCPN Wiki
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:plugins

Then https://opencpn-manuals.github.io/main/opencpn-plugins/index.html
Under Navigation  pick Celestial_Navigation
https://opencpn-manuals.github.io/main/celestial_navigation/index.html