OpenCPN/OpenCPN #4326: UI guidelines with Gnome HIG

nocodehummel, 2025-01-21:
After a discussion with contributors (#4323) the OpenCPN maintainers approved the use of [Gnome HIG](https://developer.gnome.org/hig/) as standard for UI design.  The standard is intended to be used for Gnome and not platform agnostic. Still HIG is considered useful to guide OpenCPN UI/UX design decisions. This enhancement request is intended for the helicopter perspective on improving user-experience and applying UI guidelines. 

**OpenCPN is used a wide range of target platforms and devices. The UI guidelines need to be applied in that context.**

The [Developer manuals](https://opencpn-manuals.github.io/main/opencpn-dev/index.html) will be updated to include a section or page on **UI design guidelines**. With the aim to educate on the UI guidelines, how to apply them, and what resources are available.

The other goal of this enhancement request is to develop re-usable **OpenCPN UI components** that implement the UI standard. The UI components should encapsulate the standard implementation and allow for efficient re-use. This should result in more consistency in the user-experience. For most UI components separate issues will be created to define/refine the specifications.

There will be an impact on existing dialogs to adhere to UI standards. UI components only add value when being used in dialogs. Each PR should therefor also include (partial) refactoring. However in compliance with the general guidelines for impact and sizes of changes!

Other UI design related issues can also be registered (linked) as sub-issue to this enhancement request.

This enhancement request does intentionally not specify guidelines. Guidelines and UI design decision shall be finalized in the development manual guide and the UI components. 

Hereby a link to the [Gnome HIG Design Principles](https://developer.gnome.org/hig/principles.html) as high level guidance for UI design.

leamas, 2025-01-21:
If you have problems creating sub-issues please just file a "normal" and add a note in it  that it belongs here. We will then make it a sub-issue.

nocodehummel, 2025-01-21:
I was thinking if it makes sense to create a components folder for UI components which makes it easier to identify them. But really have no idea about the possible impact on build script for the various target platforms. Would it be worth it?

nocodehummel, 2025-01-22:
What is the coding guideline for copyright statements? Some sources contain a statement others don't.

leamas, 2025-01-22:
> I was thinking if it makes sense to create a components folder for UI components which makes it easier to identify them

The natural choice would be to create a new thing under libs. This should ensure that it is created without dependencies on the core and would eventually also make it possible to add to opencpn-libs to benefit plugins.

nocodehummel, 2025-01-22:
Sounds good I will move the components to libs.

leamas, 2025-01-22:
> What is the coding guideline for copyright statements? Some sources contain a statement others don't.

Copyright law is pretty clear, we should add copyright statements to all new code. Old code is more complicated, you can't really add a statement referring to someone else than yourself.

When modifying existing sources with an existing statement, add your own claim under the existing one(s). See for  example _model/cmake/FindGLIB.cmake_ . Otherwise, don't mess with other people's statement.

leamas, 2025-02-13:
:)