OpenCPN/OpenCPN #4327: Optional GUI parts are not consistent

leamas, 2025-01-21:
**Describe the bug**
In various parts of the interface we parts which are by default hidden, often labeled "Advanced" or so. The implementation of this varies, and the user experience is thus somewhat messy

**Expected behavior**
A  common implementation with a consistent interface.

**Screenshots**
Connection properties
[image]

[image]

Connection Edit
[image]

[image]

Udev rule management
[image]

[image]

**Additional context**
The first and last one looks similar. However, the first is using UTF symbols whereas the last uses icons. The latter translates to problems when using other color schemes than the default.

bdbcat, 2025-03-24:
Anything else specific and actionable here?

leamas, 2025-03-24:
> Anything else specific and actionable here?

hm...  "Here" is just the fact that we have confusing GUI elements for show/hide gui parts typically percieved as "advanced", etc. At least the triangle and the more.../less.. labels. No more and no less, certainly possible to fix.

Low priority, could easily slip to next release.

bdbcat, 2025-03-24:
What fix do you see?  Buttons with icons?  Or...

leamas, 2025-03-24:
Using the triangles everywhere is my first thought. Probably means we create some kind of reusable stuff rather than just copy-paste. Personally, I don't have time at the moment.

rgleason, 2025-03-24:
Tesselated buttons?

leamas, 2025-05-02:
Fixed in 6db03a88a, ff0b1715f, 6db03a88a, and 283c4d9f5. 

Uses the gnome expand/collapse icons:  [image] and  [image]. Updating these is basically oneliners

Closing

closed 2025-05-02 by leamas