OpenCPN/OpenCPN #4328: Conflict in the coding guidelines to use standard string.

nocodehummel, 2025-01-22:
The coding guidelines have a conflict between the use of `std::string` or `std::wstring` and using `_()` that produces `wxString`. 

A new internationalization macro is required to resolve this issue (e.g. `i18n()`). The output of the macro should be `std::wstring`. Otherwise it is not possible to adhere to this coding guideline and phase-out `wxString`.

leamas, 2025-01-22:
> A new internationalization macro is required to resolve this issue (e.g. i18n()). 

Or, we could redefine _(), let it bomb and fix the issues. The _() macro is well established, to have a non-working variant will be confusing.

Furthermore, the redefinition could be done on file basis, to avoid a global break down.

nocodehummel, 2025-01-22:
I think it will  be problematic. Overloading only works on input variable so the compiler cannot differentiate two version based on the required output as input for a next method. In a source file there will most often be a need for `wxString` to be passed to *older* methods. And some new methods have `std::wstring` as i18n text input (e.g. a message dialog). I expect that there will often be both within the same source code. 

IMO working with two definitions for the same thing in the code base is going to get messy. Better to define a replacement and use the internationalization macro that gives the right output variable to be passed to the next method. 

```
#include <wx/intl.h>
#define i18n(str) (std::wstring(wxGetTranslation(str).wc_str()))
```

leamas, 2025-01-22:
> In a source file there will most often be a need for wxString to be passed to older methods

In almost all cases this works. That is `void method(const wxString& s)`  in a header   will compile something like
```
std::string s;
method(s);
```

works, there are `wxString(const std::string& s)` constructors. It breaks down when the argument is non-const, but it's not common with _()  results.

But we will have to make a test....

nocodehummel, 2025-01-22:
Sorry, but I still think that proposal sounds complicated and has risks. To make it work the developer needs to make sure that `_()` is redefined in a specific source. And than make sure that any `wxString` variables are properly converted. I think it is safer to define a new macro to gradually phase-out the old 5000x usage. Less impact and less risk for failure.

leamas, 2025-01-22:
>  To make it work the developer needs to make sure that _() is redefined in a specific source

Which then is about including a specific header. Not overwhelming.

> I think it is safer to define a new macro to gradually phase-out the old 5000x usage.

If this is required we need to find some other solution. Using something else than _() for, well, _() will create a horrible legacy.

nohal, 2025-01-22:
What actual problem are we trying to fix here if I may ask? I really don't remember the underscore macro causing any problems to me, ever. `wxString` is annoying in some situations, less so in recent wxWidgets making it `std::string` compatible, but really not in this context.

nocodehummel, 2025-01-22:
The problem is that the coding guidelines state we should is `std::string` or `std::wstring` instead of `wxstring`. But we do not get ride of `wxString` because the `_()` macro keeps producing them. Anything that uses translation introduces more `wxString`.

How should I follow the guideline?

transmitterdan, 2025-01-22:
I thought wxWidgets said to stop using _().

nohal, 2025-01-22:
And all the GUI uses `wxString`, are we also going to reimplement wxWidgets without it? Sorry, but what is the real problem, except "the guideline that is impossible to follow"?

nocodehummel, 2025-01-22:
@nohal there is no real problem, except for a contributor trying to follow the coding guidelines. Should I not?

nocodehummel, 2025-01-22:
> And all the GUI uses wxString, are we also going to reimplement wxWidgets without it? 

@nohal that would have been a good question when the coding guidelines were agreed.

nohal, 2025-01-22:
IMHO exchanging the well established underscore macro, which does work as is and causes no problems for the sole reason of being exactly able to say I follow the guideline a price far to high to pay and totally unneeded unless we get rid of all the dependencies on wxWidgets in the respective code.
And yes, the coding guideline is missing a "where reasonably possible".

transmitterdan, 2025-01-22:
This recommendation to prefer std::string by developers comes from wxWidgets developers as I recall. They are trying to work away from using proprietary container and string classes within wxWidgets. It will take time to get there but there really is no need to use wxString in developer variables.  As wxWidgets matures, the wasted cpu cycles will go down if developers stop creating wxString variables. At least that's what i think this is about.

nocodehummel, 2025-01-22:
What is wrong with introducing a macro that is compliant with the coding guidelines to be able to write new code with `std::wstring`? Small impact and clear solution that prevents an ever increasing usage of `wxString`. Or am i missing something?

```
#include <wx/intl.h>
#define i18n(str) (std::wstring(wxGetTranslation(str).wc_str()))
```

leamas, 2025-01-22:
One thing is here is that the coding guidelines are clearly broken, of course we need to use wxString when interfacing wxWidgets. I reviewed that PR, so my bad.  Have committed a fix, it will take some time until the manual is rebuilt.

Fully agree with @transmitterdan. 

Here is also the problems described in #4262, basically making it hard to move to C++20 with _()  producing wxString. Probably a symptom which will get worse over time as type checking is enhanced. 

> What is wrong with introducing a macro that is compliant with the coding guideline

_() just has to much mindshare. A working environment should have working _().  Nothing more complicated than that.

transmitterdan, 2025-01-22:
I think this will all become simpler when wxWidgets moves to newer version.  However, also, many things will break when wxWidgets adopts much of std::  under the covers. If you ask me, we should now be testing using latest wxWidgets and fixing what we know will break than arguing over a macro that works with no problems.

nohal, 2025-01-22:
The underscore macro is a perfect example of something that will fix itself as `wxString` becomes `std::string`, which we know is going to happen. I really don't think this is worth any investment at all.

leamas, 2025-01-22:
> What is wrong with introducing a macro that is compliant with the coding guideline

... and, remember that this works:
```
std::string s = _("foobar");
```

which means that you can use the macro without adding more wxString usage in most cases. There are corner cases, but they are indeed corner cases. And, untested, there should also be `_("foobar").c_str()` and `_("foobar").ToStdString()`

>>    However, also, many things will break when wxWidgets adopts much of std:: under the covers.

> We can prevent that for any new code using an alternative macro. Maybe _std() is a better name.

This is a much bigger issue, _() is a small part of it if any. And no, we should not add an alternatibne

nocodehummel, 2025-01-22:
@leamas it is my understanding that the result cannot be `std::string` for the supported languages and must be `std::wstring`.

Given the resistance to a macro, I will go with the in place conversion and create new interface with `std::wstring` for multi-language support.

closed 2025-01-22 by nocodehummel

leamas, 2025-01-22:
> @leamas it is my understanding that the result cannot be std::string for the supported languages and must be std::wstring.

Well, we have been using std::string for quite some time for all locales. Should not be a problem. I suggest that you adopt to how things are done today in this case.

nocodehummel, 2025-01-22:
> I suggest that you adopt to how things are done today in this case.

You mean to use `wxString` and ignore the standard or `std::string` for new methods?

leamas, 2025-01-22:
I mean using std::string in the public interface, using wxString when you interface wxWidgets internally

nocodehummel, 2025-01-22:
@leamas sorry to bother you again, but I checked with ChatGPT to translate "Delete" into Japanese.  It renders code with `std::wstring`. Below the explanation.

> When you need to handle text that may include characters from languages with non-Latin alphabets or symbols, such as Japanese, Chinese, Arabic, etc.

Are you sure `std::string` is sufficient? I do not have the experience with supporting the locales that OpenCPN does.

leamas, 2025-01-22:
Just do it. We are using utf-8 internally.