OpenCPN/OpenCPN #4329: Confirmation dialog UI component

nocodehummel, 2025-01-22:
Implement confirmation dialog UI component following as PoC for introduction of UI guidelines. And replace the route delete confirmation dialog.

[image]

Hakansv, 2025-01-22:
I'm sorry but what's the difference to the present? The same info and user actions.

[image]

nocodehummel, 2025-01-22:
@Hakansv what do you mean?

Hakansv, 2025-01-22:
Ahaa. I thought your where showing an example of some new kind of dialog to replace the present. 
So, what's your concerns' about the present dialog?

nocodehummel, 2025-01-22:
@Hakansv you asked yesterday about the concern in #4323.

nocodehummel, 2025-01-22:
@leamas can you add this as sub-issue?

nocodehummel, 2025-01-25:
Resolved in #4332.

closed 2025-01-25 by nocodehummel