OpenCPN/OpenCPN #4331: Measurement Tool only provides metric units

torrmundi, 2025-01-22:
The Measurement Tool units for small distances is in metric units regardless of other settings. 
It would be good to have it respect the choice made in Options>Display>Units: Distance.

Given the choices for Distance, the Measurement Tool should have:
Kilometers: m (small) to km (large)
Statute Miles: **should be feet (small)** to miles (large). However, small distances are currently only in meters.
Nautical Miles: **should be feet (small)** to nautical miles (large). However, small distances are currently  only in meters.
Meters: m (small and large)

rgleason, 2025-01-22:
Yes, Yards and Feet would be very useful

svMigration, 2025-04-24:
I requested this years ago on flyspray.  So adding another voice in favor here. Would like to see it change to feet where it now changes to meters  which is currently < .1 NM.