OpenCPN/OpenCPN #4333: Convert main Options dialog mode to fully non-modal

bdbcat, 2025-01-24:
Various UI elements could become more clear and intuitive if the main Options dialog was implemented as a non-modal floating dialog window.

This implementation is presently underway.
This Issue is created as a container to record any functional issues related to the non-modal settings dialog discovered during testing.

bdbcat, 2025-01-24:
Initial implementation  complete.
be4aaff6b67968fec389086328e54012f633c79c

bdbcat, 2025-01-24:
Please exercise and test all dusty corners of the options dialog.
TIA for the feedback.

Hakansv, 2025-01-24:
Tested on Win10, display scale =175%
In general very good. The options dialog is always "above" O canvas and "below" its own sub-dialogs. 
Comments regarding UI for some dialogs. (Possible noted elsewhere but to get it in one place)

(I've some functional requests but that's another issue I suppose?)

1. Page Connect-> Connections list. 
The vertical scroll bar does not work inside the connections list. To scroll, the pointer must be moved to the empty space between the list and the vertical slider. 
btw: Why that "big" empty space between the connections list and the vertical scroll bar?

2. Page Connect-> "Advanced" fold down
If the Advanced section is folded down when the Options dialog isn't large enough to cover all the content, the lower sections are hidden. It is likely that e.g. "Adjust communication priorities" will not be visible to many users since no vertical scrollbar is created. There is no reason for a user to suspect that there are more to see if the Options dialog is enlarged.

A shot to show

[image]

bcn2, 2025-01-24:
On macOS looks fine. No hickups at the moment using the latest universial version from this morning 11888.60468c025

Hakansv, 2025-01-24:
Another comment/consideration.
As can be seen on my above shot is the Option's "OK" button is replaced by "Close"
We had the function to "Apply" changes or "OK" changes. The latter also implied an apply action. That was so to say clear for the user.
Now it reads "Close" instead but I suppose it's still the same function; Close implies also a "Apply" action. But "Close" is not as clear as "OK" that changes are saved. Can the user be doubtful if "Close" really saves changed settings?
In that case we might keep "OK" instead of "Close"?

bdbcat, 2025-01-24:
An Area of concern is this:
With locale/language changes, and also with font changes, the dialog must be entirely rebuilt to apply these change.  This creates a bit of a lag in the user experience, especially when pushing "Apply" on language change.
We may instead choose to always use the "Close" code path on language change, so that this unexpected re-appearance of the options dialog does not happen.
Please try this, and let me know your thoughts.

Hakansv, 2025-01-24:
I'm not sure I fully understand the "Area of concern" and the "_"Close" code path on language change_" part.
But still on basic understanding and use of the "OK" versus "Close":
Many dialogs that includes editable preferences, like our MUI (hamburger) menu, works in the way that when any options is changed it's also directly implemented. Then a "Close" of the dialog is relevant since nothing more will happen after that.
In "Options" there are still preferences that will not be updated until the user click either of "Apply or (former) "OK". The "OK can be understood as "OK, adapt these changes". My concern is that "Close" can be misunderstood as something similar to "Cancel".  But it can be I'm overreacting? Others will have to judge it.

bcn2, 2025-01-25:
We have here three options around our two actions: 
- Cancel: don't _apply_ & _close_  
- Apply: _apply_ & _don't close_
- Ok: _apply_ & _close_

As _close_ is involved in all three options, I would prefer not to name one Close, instead staying with Ok

Hakansv, 2025-01-25:
The User dialog: 
When changing any font or language the Options dialog is "reloaded".  During that process the second Canvas is also reloaded and "destroyed" . The MUI bar is deleted. See shot. 
This is valid both for Windows and Debian.

There was also a discussion if it's possible to avoid this "reload" when fonts for anything else than the "Dialog" is changed?

[image]

transmitterdan, 2025-01-25:
> "Close" vs "OK" is a question I wrestled with. I'd like more opinions on this, considering other modern apps and user patterns on the various platforms. There seems to be no consistent approach in the UI style guides I have read. Thoughts?

What if we did this?
```
diff --git a/gui/src/options.cpp b/gui/src/options.cpp
index d9c01622c..f947d4e93 100644
--- a/gui/src/options.cpp
+++ b/gui/src/options.cpp
@@ -5781,12 +5781,15 @@ void options::CreateControls(void) {

   m_OKButton = new wxButton(itemDialog1, xID_OK, _("Close"));
   m_OKButton->SetDefault();
+  m_OKButton->SetToolTip(_("Apply Changes & Close"));
   buttons->Add(m_OKButton, 0, wxALIGN_CENTER_VERTICAL | wxALL, border_size);

   m_CancelButton = new wxButton(itemDialog1, wxID_CANCEL, _("Cancel"));
+  m_CancelButton->SetToolTip(_("Ignore Changes & Close"));
   buttons->Add(m_CancelButton, 0, wxALIGN_CENTER_VERTICAL | wxALL, border_size);

   m_ApplyButton = new wxButton(itemDialog1, ID_APPLY, _("Apply"));
+  m_ApplyButton->SetToolTip(_("Apply Changes"));
   buttons->Add(m_ApplyButton, 0, wxALIGN_CENTER_VERTICAL | wxALL, border_size);

   m_pageDisplay = CreatePanel(_("Display"));
```

Hakansv, 2025-01-25:
If we go for "Close" or "OK" the above tooltips are really good in any case.

transmitterdan, 2025-01-25:
Unfortunately, some changes within Options dialog (e.g. Connections) take effect immediately and so it's impossible to have a tooltip that is 100% correct.

Hakansv, 2025-01-25:
The Options dialog is reloaded also when changing Canvas mode. That might confuse users a lot. Is that necessary?
Actions to reproduce:
Options->Display->General. Change "Canvas Layout" - "Close"
The Options dialog box closes and the new layout is established. Then the options open again, for no understandable reason.

rgleason, 2025-01-25:
Connections Dialog (applied immediately)  - correct
> Unfortunately, some changes within Options dialog (e.g. Connections) take effect immediately and so it's impossible to have a tooltip that is 100% correct.

With buttons  "OK"  "Cancel"  "Apply"
"OK"  - Perhaps does, but "Close" might be better
"Apply"  - Does not make sense in this context.
"Cancel" - should at least "Disable" the new connection.

Perhaps this Option - Connections Dialog should just have a "Close" at the bottom?

transmitterdan, 2025-01-25:
> The Options dialog is reloaded also when changing Canvas mode. That might confuse users a lot. Is that necessary? Actions to reproduce: Options->Display->General. Change "Canvas Layout" - "Close" The Options dialog box closes and the new layout is established. Then the options open again, for no understandable reason.
> 
> And: The second Canvas is incomplete in the same way as when changing e.g. font as described above, No MUI bar is created. One have to restart O to get the MUI bar.

It's worse than that.  I can make O crash by switching back to single canvas, close options, options pops back up, close again->CRASH.  (Windows 11 - latest Master)

bdbcat, 2025-01-25:
Ok, first distillation:
1.  There is indeed no reason _ever_ to reload the options dialog when "Close/OK"  is pressed.  That's a bug, specific to canvas layout change.
2. Acknowledge the bug when switch from one to two canvas, second canvas missing MUI bar.
3. Opinions vary, but generally it sounds like "OK" is more intuitive than "Close".  In either case, the code logic does not change.  Just the label on the button.
4. There is no plan to create an "Undo" function, with related action history.
5. Confirm that "Cancel" button does not revert any changes made to connections.  It has always been so, significant effort required to implement that.  Will entertain separate RFE issue, if anyone so inclined.
6. Will investigate idea to avoid reload on font change, when font changes do not involve the element "Dialog".

leamas, 2025-01-25:
> Confirm that "Cancel" button does not revert any changes made to connections. It has always been so, significant effort required to implement that. Will entertain separate RFE issue, if anyone so inclined.

Partly true for the Edit Connection page. However, the main Connections page has working Apply/Cancel/Ok semantics. When user modifies a value in the UI nothing really happens. Cancel  basically just drops whatever value set in the UI. Apply actually updates the actual (usually global) values.

The Edit Connectiob page actually has a bunch of problems...

bdbcat, 2025-01-25:
"main Connections page has working Apply/Cancel/Ok"
Not quite. The "enable" checkbox takes immediate effect, and is not reverted by "Cancel"

rgleason, 2025-01-26:
Whether the button is OK or CLOSE.

```
m_OKButton = new wxButton(itemDialog1, xID_OK, _("Close"));
   m_OKButton->SetDefault();
+  m_OKButton->SetToolTip(_("**Accept or** Apply Changes & Close"));

```
For that page you don't need CANCEL or APPLY
Just have  OK or CLOSE

bdbcat, 2025-01-26:
Tooltips are a good idea, but...
Please consider how this string will localize to, say, German.
_("Accept or Apply Changes & Close")

bdbcat, 2025-01-26:
By latest commit to github master, I have (hopefully) corrected  items 1,2,3 and 6 in the above referenced list.
Thanks for your comments.  Please keep them coming.

Hakansv, 2025-01-26:
On Win10
a)
Items 1,2,3 and 6 are confirmed. I also tried to brake it by changing font for Dialog but both Apply and OK handled it in a relevant way.

b) Tooltip:
For translation to sv-SE: 
"Accept or Apply Changes & Close" vs "Apply Changes & Close"  Both easy to translate and useful. The latter is shorter and not a fundamental difference for a user who is not a computer nerd, I would think. 

c)
The initial notes about "Options->Connections" item 1 and 2 are not directly related to the changes for this issue. Can we handle it here or should I make a new issue out of that?

leamas, 2025-01-26:
> Not quite. The "enable" checkbox takes immediate effect, and is not reverted by "Cancel"

Right. Wll fix.

rgleason, 2025-01-26:
"Apply Changes"   is different than "Accept Changes"   

According to AI the difference in meaning is not quite what I expected.  Here is what AI says

Ah, got it! In a programming context, especially when dealing with dialog boxes, the terms "Apply changes" and "Accept changes" have slightly distinct roles:

**Apply Changes:**
- **Meaning:** When you select "Apply changes" in a dialog, it means the modifications made in the dialog are implemented and saved _without necessarily closing the dialog_. This allows users to see the effects of their changes and make further adjustments if needed.
- **Example:** You might see this in settings dialogs for software applications where you can apply changes to configuration options and continue tweaking them without closing the window.

**Accept Changes (or OK):**
- **Meaning:** Selecting "Accept changes" usually means that the changes made are confirmed, saved, _and the dialog is closed._ This is often equivalent to clicking an "OK" button.
- **Example:** In a dialog window where you are configuring some settings, pressing "Accept changes" (or "OK") will save the changes and close the dialog, returning you to the main interface or whatever window was previously open.

So, if you're writing a dialog for a program and you want users to see the impact of their changes before they exit the dialog, you might use "Apply." But if you want the dialog to close and save the changes as they exit, you would have them "Accept" or click "OK."

Hakansv, 2025-01-26:
> So you could have CLOSE with a message "Accept changes".

That seems to overdo things and force another user action. Nothing has changed in this matter and it have been working well the last (10?) years.

leamas, 2025-01-26:
The meanings of the  buttons "Apply", "Ok" and "Cancel" are well established out there, it's nothing we should reinvent (and no AI is needed to sort out their meaning, for sure.)

The underlying issue is that modern GUIs tends to not use them at all. Instead, changes made in the UI are effective immediately.  However, at least the Options dialog is heavily designed around the old paradigm, and it's IMHO better to be consistent than change random pieces to the more modern style.

leamas, 2025-01-26:
> So you could have CLOSE with a message "Accept changes".

Any example of an application which uses this in the wild?

bdbcat, 2025-01-26:
"The initial notes about "Options->Connections" item 1 and 2 are not directly related to the changes for this issue. Can we handle it here or should I make a new issue out of that?"

I would prefer a new issue, please.

bdbcat, 2025-01-26:
"So you could have CLOSE with a message "Accept changes"."
Not necessary.  If you want to close _WITHOUT_ changes, there is "Cancel" available.

rgleason, 2025-01-26:
Not necessary. If you want to close WITHOUT changes, there is "Cancel" available.  
Agreed.

leamas, 2025-04-27:
The main task to make Options modeless is  completed. Time to close?

bdbcat, 2025-04-27:
Done, complete.

closed 2025-04-27 by bdbcat