OpenCPN/OpenCPN #4339: BUG 1/2: PluginManager access to cache button

rgleason, 2025-01-25:
**Describe the bug**
Once a user upgrades to a new plugin, and since the master catalog can only have a Plugin set with one version number, there is no other older Metadata files available for that plugin, and thus there is no easy way to "Downgrade" or have access to the **"tarballs"** in  opencpn/plugins/cache/tarballs.

**Solution**
Provide the uses with a  "Tarball Cache" button that will access the opencpn/plugins/cache/tarballs stored on the system, and allow them to select a tarball and install it using the "Import Plugin Tarball" procedure.

[image]

bdbcat, 2025-03-10:
This is a Feature Request.
Please move this issue to the appropriate index.

leamas, 2025-03-14:
Changed label to Enhancement

rgleason, 2025-04-24:
I still call it a 1/2 bug.  I've already had to tell users a number of times to "Uninstall" any "imported" plugins before being able to upgrade, because "upgrades" do not appear after updating the master catalog.   Indeed any upgrade is inaccessible even by version number.

Users state "There is no upgrade available."