OpenCPN/OpenCPN #4340: Segmentation fault on opening wpt property dlg

RooieDirk, 2025-01-25:
If I double-click (or right-click & choose properties) on a waypoint, I get a segfault. Opening the propdlg from the routemanager dlg works OK.
Nothing in the log, running from commandline only gives "segmentatationfault".  gdb is a little more helpfull
`Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x00007ffff776e62e in wxWindowBase::SetValidator(wxValidator const&) () from /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.2.so.0
`
If the waypoint is part of a route things work as expected. Clicking a route also works.

Running todays git master on debian 12

leamas, 2025-01-25:
Probably related to #4335 and the related discussion on https://github.com/OpenCPN/OpenCPN/commit/af1520888fe9c485290b2f0073c0278900dac279. Lets see if the PR fixes it.

nocodehummel, 2025-01-27:
@RooieDirk the above mentioned fix is merged. Does that resolve the issue?

RooieDirk, 2025-01-27:
Yes it does. Thanks.

closed 2025-01-27 by RooieDirk