OpenCPN/OpenCPN #4343: Options->Connections dialog. Scroll bar obstacle for source list and Advanced content is hidden

Hakansv, 2025-01-26:
**Describe the bug**
In Options->Connections
Two issues
1. Page Connect-> Connections list. 
The vertical scroll bar does not work inside the connections list. To scroll, the pointer must be moved to the empty space between the list and the vertical slider. 
btw: Why that "big" empty space between the connections list and the vertical scroll bar?

2. Page Connect-> "Advanced" fold down
If the Advanced section is folded down when the Options dialog isn't large enough to cover all the content, the lower sections are hidden. It is likely that e.g. "Adjust communication priorities" will not be visible to many users since no vertical scrollbar is created. There is no reason for a user to suspect that there are more to see if the Options dialog is enlarged.
See shot below.

**To Reproduce**
Steps to reproduce the behavior:
1. See 1. and 2. above

**Expected behavior**
The bug-part is 2. above.
The Advanced section should be possible to view on all screens and dialog size.

**Screenshots**
A shot to show the dialog when "Advanced" is viewed.

[image]

**Desktop
 - OS: Win10 and Debian RPi4 & 5

**Additional context**
Also mentioned here #4333 but as requested extracted to this issue.

Hakansv, 2025-02-10:
The partial fix confirmed. Now scrolling works also in the connections frame. Thanks

Sorry for two bug reports in the same issue. Will try to avoid it in the future.

Hakansv, 2025-03-27:
For the sometimes invisible "Adjust priorities" button.
My intention is to move that button so that it is visible at all times and not hidden in "Advanced".

An important reason for this is partly that after "Add new connection" it is very possible that the priority will change. First, that a somewhat inexperienced user may need a hint that the prioritization feature exists and perhaps should be used or investigated. When the priority is hidden in Advanced, many users probably miss this. 

I I also propose a minor text amendment for the control to clarify its use. (Native language adaptation welcome if appropriate :-) )

A PR giving this look is underway, if no objections here.

[image]

bdbcat, 2025-03-27:
No objections here.

commit 109775c4 (2025-03-28): Merge pull request #4420 from Hakansv/conprio

Options-Connections, move Adjust priorities button  Closes #4343
commit 324ebe3b (2025-03-28): Options-Connections, move Adjust priorities button  Closes #4343
commit 9332ce60 (2025-03-31): Options-Connections, move Adjust priorities button  Closes #4343
closed 2025-03-28 by bdbcat