OpenCPN/OpenCPN #4345: BUG: Navobj Backup - Better access and tools.

rgleason, 2025-01-27:
**Describe the bug**
5hrs of work on a Planning and Presentation Routing Chart went to oblivian, just before I was going to finish and close. I edited WP properties to show the "Empty" symbol and something happened. There was a loud bleap and O crashed. I started it up again, not knowing what was wrong (a mistake). What I think happened was OpenCPN tried the navobj.xml file and found it was corrupt, so it overwrote it with the navobj.xml saved about 5 hours before that, rather than first saving the corrupt file as "**navobj.xml.corrupt**"

**To Reproduce**
I can't reproduce exactly what happened, but I was changing Mark Icons to the "Empty" Icons for a long time.

**Expected behavior**
When there is a failure of OpenCPN like that

- I expect the navobj.xml file to be copied to navobj.xml.corrupt if it is corrupted rather than simply overwriting it with the most recent navobj.xml file!
- I need this done because I don't always kknow if the navobj.xml file is involved, so I don't always check it for corruption before starting Opencpn again.
- Easily save the navobj.xml file back to the disk whenever User hits an easy to reach button.
- Set a time parameter with a User option to save to disk.
- Ability to set the name of the timed backup file to something other than navobj.xml.
- Ability to browse to select the location for these backup files. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: Win 11
 - Version5.10.2

**Additional context**
The fact that I lost 5 hours of work was avoidable with better backup tools.
We desperately need these tools, too many people report these failures, so with better backup tools there will be less disappointment.

Then there is the question about what caused it? ...Something to do with editing Waypoint Properties?  It seems to me that Hakan and others have rund into this too.

nocodehummel, 2025-01-28:
IMO the short term solution would be to also keep backups of the `navobj.xml` changes file for each session. A similar solution as is used for the actual `navobj.xml` backup. So at least the data is not lost. And potentially can be imported as GPX.

nohal, 2025-01-28:
`navobj.xml.changes` is not GPX, but a WAL style record of changes that is replayed on top of `navobj.xml`, which is GPX. All we need is not to throw it away if it is unusable and finally afer all these years we can't reproduce this actually see what is the problem...

rgleason, 2025-01-28:
So it was a file navobj.xml.changes  (WAL style record of changes) that crashed?
Leaving a 4-5 hour old navobj.xml file!!

There have to be much better ways to handle this. It is an Old problem that has been solved.  I think the navobj.xml file is fine, some people think should all be in a database, but plugins use the navobj.xml file in different ways. I think the crash problems stem from the changes file and how that is handled. It gets overloaded or something and needs to be flushed out regularly.

Our CAD program does a wonderful job of recovery, every time it opens a drawing file it checks the file, and it checks to see if there was a crash leaving a corrupted file. It fixes the file, removing the bad entity and reloads it. From there you can see what happened and decide to continue or go back to an earlier version.

Why not have the navobj.xml.change file be "replayed on top of navobj.xml, which is GPX" every 20 minutes or when the user wants it "replayed" by having an actual Button that they can hit whenever they want to save the navobj.xml?

What is the problem? ---Is it that once you hit the button or save it every X minutes it renames all the backup xml files and pushes them up the line, so if a file is corrupted the older copies are not available?

nohal, 2025-01-28:
A file on disk can not crash, it is OpenCPN that crashes and I have never seen it crash while writing this file. `navobj.xml.changes` is used in append-only mode, so it should never become unusable.
I have not been able to cause a crash that makes it unusable for years and believe me, I tried hard.
I need to see `navobj.xml.changes` that is not replayable to come to any conclusions. The only thing I am interested here in is that specimen.

rgleason, 2025-01-28:
I want to know why 

1. It went 5 hours without saving the changes?
2. Why can't we have timed parameter for saving the changes?
3. Why can't we have a "Save Navobj" Button up front which allows me to save the changes?

Where do I find this **navobj.xml.changes** file and how do I catch it?  You want it only after it has failed and disappeared, so it appears you are asking for the impossible.

Those things would help us help you.

nohal, 2025-01-28:
If you want to help, next time you have crash, save `navobj.xml.changes` and in case it is not replayed, provide it to us together with the actual `navobj.xml`. That is the only relevant information here.

nocodehummel, 2025-01-28:
Are the `navobj.xml.changes` always replayed when restarted after a crash? Does it matter if the user selects *Safe mode*?

rgleason, 2025-01-28:
No answers to the other questions.
OK, I will add that to a section of FAQ and post it in the forum.

I have a request that when OpenCPN is restarted, and it finds a navobj.xml.changes file that it copy that file to navob.xml.changes.bkup before continuing.  Then you'll have a better chance of getting these files.

rgleason, 2025-01-28:
This is a black box that fails for everyone once in a while.

nohal, 2025-01-28:
Yes, that is of course on the list and has been requested even here already, no need to repeat the obvious.

nocodehummel, 2025-01-28:
Did a small test on the changes file. This file is created when I start OpenCPN. Is the below content not usable as GPX?

[image]

```
<wpt lat="50.350706798" lon="-0.491470696">
 <time>2025-01-28T13:52:17Z</time>
 <sym>activepoint</sym>
 <type>WPT</type>
 <extensions>
  <opencpn:guid>17410000-85bf-4afe-a66c-02387fafffff</opencpn:guid>
  <opencpn:arrival_radius>0.050</opencpn:arrival_radius>
  <opencpn:waypoint_range_rings visible="false" number="0" step="1" units="0" colour="#FF0000" />
  <opencpn:scale_min_max UseScale="false" ScaleMin="2147483646" ScaleMax="0" />
  <opencpn:action>add</opencpn:action>
 </extensions>
</wpt>
<rte>
 <extensions>
  <opencpn:guid>19600000-58ee-42d4-982a-e8575cd9ffff</opencpn:guid>
  <opencpn:viz>1</opencpn:viz>
  <opencpn:planned_speed>6.00</opencpn:planned_speed>
  <opencpn:time_display>PC</opencpn:time_display>
  <gpxx:RouteExtension>
   <gpxx:IsAutoNamed>false</gpxx:IsAutoNamed>
  </gpxx:RouteExtension>
  <opencpn:action>add</opencpn:action>
 </extensions>
 <rtept lat="50.077061101" lon="-0.613649007">
  <time>2025-01-28T13:54:18Z</time>
  <name>001</name>
  <sym>diamond</sym>
  <type>WPT</type>
  <extensions>
   <opencpn:guid>6e1fffff-f71f-4bcf-97ac-0a4a4544ffff</opencpn:guid>
   <opencpn:auto_name>1</opencpn:auto_name>
   <opencpn:arrival_radius>0.050</opencpn:arrival_radius>
   <opencpn:waypoint_range_rings visible="false" number="0" step="1" units="0" colour="#FF0000" />
   <opencpn:scale_min_max UseScale="false" ScaleMin="2147483646" ScaleMax="0" />
  </extensions>
 </rtept>
 <rtept lat="50.151241710" lon="-0.076844302">
  <time>2025-01-28T13:54:18Z</time>
  <name>002</name>
  <sym>diamond</sym>
  <type>WPT</type>
  <extensions>
   <opencpn:guid>11370000-71bc-4027-8a09-f79b659effff</opencpn:guid>
   <opencpn:auto_name>1</opencpn:auto_name>
   <opencpn:arrival_radius>0.050</opencpn:arrival_radius>
   <opencpn:waypoint_range_rings visible="false" number="0" step="1" units="0" colour="#FF0000" />
   <opencpn:scale_min_max UseScale="false" ScaleMin="2147483646" ScaleMax="0" />
  </extensions>
 </rtept>
 <rtept lat="50.192865800" lon="0.357278631">
  <time>2025-01-28T13:54:19Z</time>
  <name>003</name>
  <sym>diamond</sym>
  <type>WPT</type>
  <extensions>
   <opencpn:guid>5de6ffff-616e-44fb-8a22-7da67dacffff</opencpn:guid>
   <opencpn:auto_name>1</opencpn:auto_name>
   <opencpn:arrival_radius>0.050</opencpn:arrival_radius>
   <opencpn:waypoint_range_rings visible="false" number="0" step="1" units="0" colour="#FF0000" />
   <opencpn:scale_min_max UseScale="false" ScaleMin="2147483646" ScaleMax="0" />
  </extensions>
 </rtept>
</rte>
```

rgleason, 2025-01-28:
Is the only time that the append only mode navobj.xml.changes file can be "overlaid" onto the navobj.xml file, when OpenCPN is closing?

bdbcat, 2025-01-28:
On Android, we merge the changes file, and flush the navobj.xml, every 5 minutes.  We do it like this on Android because we want to protect the user from having to do any file system activities manually.  A true pain on Android.

See ocpn_frame,cpp:5822

```
  // Update the navobj file on a fixed schedule (5 minutes)
  // This will do nothing if the navobj.changes file is empty and clean

```
No reason not to do this on all platforms, now that we have advanced to reasonably good performance everywhere.  Might debate that this will unnecessarily stress SSD write cycles, but probably moot point since Windows is hammering on disk so much anyway.

tl;dr
@nohal:  Please require the code snippet and give me any thoughts/improvements.

rgleason, 2025-01-28:
You'll see a lot more of the navobj.xml files if you implement this:

> ! have a request that when OpenCPN is restarted, and it finds a navobj.xml.changes file that it copy that file to **navob.xml.changes.bkup** before continuing. Then you'll have a better chance of getting these files.

rgleason, 2025-03-22:
Let's hope we can get an "unusable"  navobj.changes file from someone,  for Pavel!

rgleason, 2025-03-28:
@nohal   I've got one for you!

https://github.com/rgleason/weather_routing_pi/issues/238

rgleason, 2025-03-28:
Got it to fail again and produce another navobj.change.
https://github.com/rgleason/weather_routing_pi/issues/238#issuecomment-2761455668

bdbcat, 2025-05-03:
Please note:
https://github.com/OpenCPN/OpenCPN/issues/4201#issuecomment-2848382721

Hakansv, 2025-05-03:
Win10
The new track data base function works. I've tried to stress it to mismatch but so far no issues including first time creation and transfer existing tracks from navobj.xml to navobj.db.

rgleason, 2025-05-03:
https://github.com/OpenCPN/OpenCPN/commit/84f2b70bf9e5d825a27d753a6fc4464c45b2dc04

bdbcat, 2025-05-06:
Yes, start over if necessary

rgleason, 2025-05-06:
Thanks Dave

closed 2025-05-06 by rgleason