OpenCPN/OpenCPN #4351: New connection dialog: Signalk "driver not avaiable""

Hakansv, 2025-02-01:
**Describe the bug**
A Signalk connection won't display "Data flowing" status but instead "No driver statistics available" although data is flowing normally.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a SK data source 
2. Study status -  "No driver..."
3. Open NMEA Debug window and see that data is actually flowing.

**Expected behavior**
A relevant status flag.

**Screenshots**
See below

**Desktop (please complete the following information if applicable):**
 - OS: Win & Debian(Rpi)

** Test to verify**
You may use signalk.stupan.se:3000 to test if no own source is available

[image]

leamas, 2025-02-01:
This is not really a bug. The message is correct implying that patches to the signalK driver to ptovide stats are yet not in place.

Hakansv, 2025-02-01:
Ok, on the "list" with other words. Unless we don't need a reminder I'll then close this issue?

leamas, 2025-02-01:
Or make it an RFE?

leamas, 2025-02-01:
Btw: help welcome. I:m ready to assist...

Hakansv, 2025-03-18:
Corrected by 291628b9a . Thanks.

closed 2025-03-18 by Hakansv