OpenCPN/OpenCPN #4353: OpenCPN Warning: Malware / spyware version?

apezio, 2025-02-04:
Yesterday, an OS warning/notification window popped up saying OpenCPN (5.8.4-0+1637c28) was requesting access to view or record my screen!?  WTF big red alert in my mind.  I immediately zipped the .app (it makes a copy) and deleted the executable version.... haven't done anything else since.

I recently migrated from OSX 10.14 to 15.x (Sequoia). In 10.14 the OS has no such warning, but Sequoia has a lot more limitations on software and more notices for this sort of thing...

I was not using OpenCPN recently nor was OpenCPN running at the time.

the md5 checksum for the binary I have, is: 
b92468b3191f1d39f80bf14d1bd89f72  OpenCPN_5.8.4.app/Contents/MacOS/OpenCPN

I do not know where I downloaded it, but would guess it was via Github.   I checked just nbow, and the version on Github has the same md5.

I should have taken a screenshot, but I am 99% sure I was not hallucinating.

bdbcat, 2025-02-04:
apezio...
You are not hallucinating.  This happens from time to time on modern MacOS releases, even when using the official github Releases of OpenCPN.  It is a known issue.
We judge this error to be a "false positive" warning.  OpenCPN does not, nor has never, captured or recorded screen contents, on any platform.  A close inspection of the source code will verify this.
Apple provide no information on their intrusion detection logic, for obvious reasons.
So we, and other trusted apps, (google this), have learned to live with it.

Thanks
Dave

apezio, 2025-02-04:
Good to know I'm probably not hacked and that OpenCPN is probably not compromised.

I don't know how many lines of code OpenCPN has, but if it's more than a ~10, I don't think I can verify that for myself.  

Openssls public source code was backdoored by a developer not long ago and only one person noticed, so I'm kinda paranoid!

Thanks for the info.

leamas, 2025-04-09:
Anyway, we have actually no bug here. Closing.

Thanks for reporting!

closed 2025-04-09 by leamas