OpenCPN/OpenCPN #4356: Flatpak OpenCPN doesn't run on new 16Gb Pi 5

jimseng, 2025-02-06:
Just received a Pi 5 16Gb version and flatpak OpenCPN comes up with a blue/ striped screen and then the whole screen hangs, sometimes with a movable mouse cursor, sometimes completely frozen. Rpi still responds to login via SSH but all that is left is to do a reboot, which can take a couple of minutes to achieve. This is with both Openplotter version and fresh 64 bit bookworm install. Swapping brand new bookworm install to a Pi5 8Gb (i.e. same fresh image and flatpak install on same USB stick) and there are no issues.

leamas, 2025-02-10:
There seems to be general problems with the Rpi5 16G. See for example https://discourse.ubuntu.com/t/ubuntu-24-04-r-pi-problems-with-the-new-16gb-pi-5/53051

jimseng, 2025-02-10:
Yes, it seems specific to flatpak. I only used the flatpak version to get two finger zoom  working with my touch screen. Luckily there is now a solution that I have since discovered which I started to discuss here, but it looks like it comes with its own issues: https://www.cruisersforum.com/forums/f134/pi5-16gb-flatpak-touch-screen-pinch-zoom-wayland-good-news-292331.html
And I have given up on wayland!

leamas, 2025-02-10:
You might want to talk to the Flatpak people to see if they have any answers: https://lists.freedesktop.org/mailman/listinfo/flatpak or https://discourse.flathub.org/. I have had good response on some issues on the mailing list.

leamas, 2025-04-27:
The Ubuntu problem seems to be fixed by a snap mesa update.

Since this bug was reported we have also updated the Flatpak runtime which means the bug **could** be fixed. However, I have no Rpi-5/16Gb at hand.

Let's see if we can get this tested during 5.12 beta cycle

jimseng, 2025-04-27:
Not that I use the flatpak version anymore but I just installed the beta version on my Pi 16 and it seems to works fine now.

leamas, 2025-04-27:
Thanks for testing! Can we agree to close this one? If so, please do, our policy is that OP closes if possible.

jimseng, 2025-04-27:
Yes.

leamas, 2025-04-27:
OK. Thanks for reporting and not least testing!

closed 2025-04-27 by leamas