OpenCPN/OpenCPN #4362: Constructor ocpnDC can fail if initializer DC does not contain valid bitmsp

transmitterdan, 2025-02-09:
I think this line should be as follows:
```
  if (pmdc && pmdc->GetSelectedBitmap().IsOk())
```
If the there is no selected bitmap of incoming pdc or bitmap is invalid then the wxGraphicsContext::Create() function will fail.

transmitterdan, 2025-02-10:
In looking at it a bit longer the whole if statement doesn't make a lot of sense to me. Anybody know why the if (pmdc) even exists?

bdbcat, 2025-02-10:
Looks like this convoluted process to distinguish between a wxMemoryDC and a wxClientDC is probably no longer needed, but possibly critical for some earlier wxWidgets code version.
Sure looks like the whole thing could be simplified, considering the current wxWidgets documentation.