OpenCPN/OpenCPN #4364: Fonts are not saved.

RooieDirk, 2025-02-12:
Todays git master.
If I change a font (type, size or color) then the view is updated, but the new font is not saved in the config file.
The saved font will be the default type.

If I change the font by hand in the configfile, the updated font is not used. 

This issue does at least apply to AIS target name and charttext. Didn't try them all.

My system is running debian 12

RooieDirk, 2025-02-13:
did a git bisect.
`
dirk@office:~/Development/OpenCPN$ git bisect bad
54311130db0766c9db78f23018a5f5936bc86fb6 is the first bad commit
commit 54311130db0766c9db78f23018a5f5936bc86fb6
Author: Sebastien Rosset (serosset) <serosset@cisco.com>
Date:   Mon Jan 6 15:24:57 2025 -0800

    Add code comments to FontDesc.
    Fix lookup of default font.

 gui/include/gui/FontDesc.h | 19 +++++++++++++++++++
 gui/src/FontMgr.cpp        | 45 +++++++++++++++++++++++++++++++++++----------
 2 files changed, 54 insertions(+), 10 deletions(-)
dirk@office:~/Development/OpenCPN$ 
`

RooieDirk, 2025-02-13:
Try to get my fingers around it, without much success until now. AIS seems to work, but charttext / statusbar text doesn't.
Also get double entry's in the dropdown select box. I am suspecting the templates, but I'm not sure.

[image]

Hakansv, 2025-02-13:
I think a [solution ](https://github.com/OpenCPN/OpenCPN/pull/4360#issuecomment-2647377274)is on going, at least what my W10 test shows.

sebastien-rosset, 2025-02-17:
The issue should be resolved, #4360 was merged. Can you verify?

RooieDirk, 2025-02-17:
Sebastien,

Almost. 
For me running debian bookworm on x86.. gtk3
First I delete all font settings by hand in the config file. start/stop O and the configfile is now setup with default font settings. OK
`en_US-e8c3cba54cbe3a94=ChartTexts:Noto Sans 12:rgb(0, 0, 0)`
Now change the font to droid 8pt. OK
`en_US-e8c3cba54cbe3a94=ChartTexts:Droid Sans 8:rgb(0, 0, 0)`
Now change the Charttext color to grey. Screen is updated to the right color 
`en_US-e8c3cba54cbe3a94=ChartTexts:Noto Sans 12:rgb(0, 0, 0)`
And the configfile settings are going back to default.

Changing both font & color in one go, the settings are saved OK. I can't compare how things work using a different OS
Dirk

PS It seems that the chartext fontsize also have somewhere in the code a minimum pt setting. I can't get it smaller than 10pt.

RooieDirk, 2025-02-19:
@sebastien-rosset,
I find a inconsistent use in the fontmanager between translatable and nontranslatable strings.
If I change in line 558 (FontMgr.cpp) _T("ChartTexts"), 
into: _("ChartTexts"),
I am able to change the charttext fontstyle. 
Changing charttext color works.
Changing the chartext font size is still resetting the font to default.

sebastien-rosset, 2025-02-19:
> [@sebastien-rosset](https://github.com/sebastien-rosset), I find a inconsistent use in the fontmanager between translatable and nontranslatable strings. If I change in line 558 (FontMgr.cpp) [...]

I agree in general it should use translatable strings. I checked the callers for these strings, and all of them use translatable strings, at least in OpenCPN core.
```
AISTargetAlert
AISTargetQuery
StatusBar
AIS Target Name
ObjectQuery
RouteLegInfoRollover
ExtendedTideIcon
CurrentValue
Console Legend
Console Value
AISRollover
TideCurrentGraphRollover
Marks
ChartTexts
ToolTips
GridText
```

However, some callers for `Dialog` and `Menu` need to be fixed. For example:
```c++
OCPNSoundPanel::OCPNSoundPanel(wxWindow* parent, wxWindowID id,
                               const wxPoint& pos, const wxSize& size,
                               wxString title, wxString checkLegend,
                               wxString selectLegend, wxString* pSoundFile)
    : wxPanel(parent, id, pos, size, wxBORDER_NONE), m_soundPlaying(false) {
  wxFont* pif = FontMgr::Get().GetFont(_T("Dialog"));
  SetFont(*pif);
```

AFAIK, the "Fonts are not saved" issue has been fixed. IMHO, this should be tracked as a separate issue.

RooieDirk, 2025-02-20:
@sebastien-rosset 
Think I found what is happening. With the translation solved I can change the font and color. The font size only very limited. Closing O saves the font OK. Opening again the font style is not respected, color is. Closing and opening a second time the charttexts font is back to all default. This is due to code in the S52plib.cpp.  I will see what I can do there.

Do you do some work on the Translate/nonTranslate stuff or should is it for me?

sebastien-rosset, 2025-02-20:
> [@sebastien-rosset](https://github.com/sebastien-rosset) Think I found what is happening. With the translation solved I can change the font and color. The font size only very limited. Closing O [...]

I don't have a plan to work on this.

closed 2025-02-20 by RooieDirk