OpenCPN/OpenCPN #4371: Wrong standard dialog layout

leamas, 2025-02-16:
**Describe the bug**
The new standard dialog, aimed to standardize and clean up the interface, has wrong layot.

**To Reproduce**
Steps to reproduce the behavior:
1. Start opencpn
2. Make a unclean exit, for example ctrl-C after linux command line start
3. Restart opencpn
4. See error

**Expected behavior**
INstead of being centered the buttons should be placed according to platform guidelines -- on Gnome this means being right adjusted

**Screenshots**

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Linux (fedora), likely to affect all platforms in different  ways.

**Additional context**
This is most likely about the components (wxStdDialogButtonSizer, wxStaticLine) not being expanded horizontally.

leamas, 2025-02-16:
@nocodehummel ^

nocodehummel, 2025-02-16:
The restart dialog is changed by #4366 which is ready for review (navigation dialog was changed by #4344). Below a screenshot of the new dialog on Windows (af6aa81).

[image]

leamas, 2025-02-16:
Different, but still wrong. For example, on GTK wxStdDialogButtonSizer places the buttons right adjusted. Some exceptions like help buttons are on the left side, though.

See for example the hotkeys dialog (the '?' canvas hotkey):

[image]

But you can basically grep for wxStdDialogButtonSizer, it's used in a lot of places.

Actually, the root cause ie easily spotted in the code: the component is not expanded. To repeat, this component must be expanded so it "owns" the complete row where it's placed. Other, like now, it is only laid out to the minimum size and the placed somewhere in the row. Is used to be centered,  it's now  left adjusted. None of this is relevant, the component should be able to place the buttons anywhere in the row

leamas, 2025-02-16:
Easily fixed by following patch:
```
diff --git a/libs/gui/src/dialog_alert.cpp b/libs/gui/src/dialog_alert.cpp
index adaa62642..20d3710d4 100644
--- a/libs/gui/src/dialog_alert.cpp
+++ b/libs/gui/src/dialog_alert.cpp
@@ -45,9 +45,8 @@ AlertDialog::AlertDialog(wxWindow* parent, const std::string& title,
     footer->AddButton(ok_button);
     footer->Realize();
   }
-
-  m_layout->Add(footer, wxSizerFlags().Border(
-                            wxALL, GUI::GetSpacing(this, kDialogPadding)));
+  auto spacing = GUI::GetSpacing(this, kDialogPadding);
+  m_layout->Add(footer, wxSizerFlags().Border(wxALL, spacing).Expand());
 
   Bind(wxEVT_TIMER, &AlertDialog::OnTimer, this);
 }
```

nocodehummel, 2025-02-19:
The above patch is irrelevant for the **Safe Restart** dialog in the master branch as it is not an `AlertDialog`. PR #4366 changes the dialog to become a member of `AlertDialog`.

leamas, 2025-02-25:
Fixed by 92875bce4 for safe restart and 776b7fe3d for "welcome to opencpn" dialogs. Closing.

nocodehummel, 2025-03-29:
@leamas I see that this is closed after the footer issue we previously discussed was resolved. The issue is only partially resolved with the resolution for the footer. Can it be reopened to resolve the outstanding wrong standard dialog layout?

leamas, 2025-03-31:
done

nocodehummel, 2025-04-02:
Resolved with #4366 (merged).

closed 2025-04-09 by leamas