OpenCPN/OpenCPN #4376: Display bug: map keeps jumping between two positions

ocornu, 2025-02-19:
**Describe the bug**
Commit 72619ce9b93bf ("Improve 10 Hz Look-ahead mode response stability") has introduced a display bug where the map keeps jumping between two positions.

**To Reproduce**
1. Make sure "center map on boat" (or whatever this feature is called) is selected. 
2. Click on the map and move it a bit (boat must still be visible on the screen).
3. Witness map jumping from that position back to the boat, then back to that position, back to the boat, etc.

**Expected behavior**
Before this commit, the map would center back on the boat after a short delay, and remain like this.

**Screenshots**
Only a video would showcase this bug correctly.

**Desktop (please complete the following information if applicable):**
 - Ubuntu Linux

**Additional context**
I've only tested this on a still boat (boat in harbor). Bug might not trigger if the boat is moving.
As it's linked to the map centering feature, a location source (GPS) might be necessary.

ocornu, 2025-02-20:
> I have identified a basic problem here, with an easy correction. However, I cannot reproduce the "jumping" effect locally. Please confirm your display settings: OpenGL ON/OFF? Autofollow mode engaged? Lookahead engaged? North-up, or course-up mode?

Here's a [video](https://streamable.com/e4zs0d) of 72619ce running in safe mode.
Unfortunately, the mouse cannot be seen, so let me explain a bit what happens…
- In the beginning, you can see on the bottom right that "autofollow" is engaged, yet the map isn't centered nor centering.
- ~5s in, i try disengaging and re-engaging "autofollow", to no avail: map does not flinch.
- ~10s in, i try it again, the map does center and jumping begins.
- ~25s in, i move the map elsewhere to demonstrate jumping happens between whichever arbitrary location i set the map to and where the boat is located: it's unlikely related to "lookahead", which is turned on (on top of that, boat is static in harbour).

OpenGL is ON in this video, but the jumping bug is there with OpenGL OFF as well.

bdbcat, 2025-02-20:
I have corrected the basic problem, I think. Please try a build with github current master.
Let me know here how this goes, corrected or any other effects noticed.
Thanks

ocornu, 2025-02-20:
Thanks. I'm afraid [it did not work](https://streamable.com/nf4sxi) though… :-/

ocornu, 2025-02-20:
It's not "just a display bug": opencpn actually believes the boat is wherever i moved the map to, as you can see in the following picture. The equidistant circles (in grey) are centered on that location it's jumping to/from:
[image]

bdbcat, 2025-02-20:
Just to isolate the problem, please verify that behavior is as expected with "lookahead" OFF.

ocornu, 2025-02-20:
> Just to isolate the problem, please verify that behavior is as expected with "lookahead" OFF.

Behavior is slightly different with ["lookahead" turned OFF](https://streamable.com/cxzezj) (in these videos i just turn "follow-up" ON/OFF: no map moving by hand but the initial one) :
 - opencpn still confuses the arbitrary screen position with the boat position: it centers back to this arbitrary location _when i turn "autofollow" back ON_.
 - however, equidistant circles remain centered on the boat at all times.
 - there is no slow jumping back and forth _per se_, but there's a [weird behavior](https://streamable.com/9anr3q) that looks a lot like an accelerated version of the same behavior, in both "course-up" modes (but not in "north-up" mode).

bdbcat, 2025-02-20:
To clarify:  Please manually turn OFF "lookahead".  Leave it off for this test.  Then try dragging, autofollow, etc.

I think the behavior we see with lookahead OFF is as designed.  We allow auto-follow from arbitrary location on the screen, set by chart dragging.  You may clear the "arbitrary position" by dragging the chart so that ship is off screen.  Then enable autofollow again.

From the user manual:
```
 Clicking the AutoFollow button will center the chart directly on your vessel's current location and toggle the icon to be greeen, as below. OpenCPN will revert Autofollow back to OFF if Autofollow is active and Own Ship is moved outside the Chart Panel. When that happens, Autofollow will then be turned OFF. To override, just toggle the button.
If you Left Click and grab the canvas and move the boat to were you want it on the screen, and it will stay in that relative position from the chart panel's side and top.

```
In the "wierd behavior" video, please confirm that lookahead is definitely OFF.  I cannot tell, but this is what would be seen with live GPS and lookahead on while chart dragging.

ocornu, 2025-02-20:
> To clarify: Please manually turn OFF "lookahead". Leave it off for this test. Then try dragging, autofollow, etc.

To clarify: "lookahead" was turned OFF in both videos.

> I think the behavior we see with lookahead OFF is as designed. We allow auto-follow from arbitrary location on the screen, set by chart dragging. You may clear the "arbitrary position" by dragging the chart so that ship is off screen. Then enable autofollow again.

Ok, I wasn't aware of that (i always have "lookahead" ON). I assumed it would behave as it does with "lookahead" ON: jump back to the boat location.
The bug very likely lies in this different handling of an arbitrary map position between those different modes…

> In the "wierd behavior" video, please confirm that lookahead is definitely OFF. I cannot tell, but this is what would be seen with live GPS and lookahead on while chart dragging.

It was turned OFF.

bdbcat, 2025-02-20:
OK, now, after getting in exactly the same modes and screen position, I can reproduce the behavior while pushing ON and OFF the autofollow button, as shown in https://streamable.com/cxzezj
Will investigate.

ocornu, 2025-02-20:
> Ok, I wasn't aware of that (i always have "lookahead" ON). I assumed it would behave as it does with "lookahead" ON: jump back to the boat location. The bug very likely lies in this different [...]

Now that i got a bit more time to think about it…
I do understand how "autofollow" _without_ boat-centering can be useful, as a hand-tailored version of "lookahead". However, i feel it is a mistake, both logically and ergonomically, to conflate it with the "standard", "non-lookahead" mode (which should center the boat and deactivate if you drag the chart by hand).
Ergonomically speaking, I believe the convoluted way to revert back to centering the map on boat (drag it until the boat is no longer visible, then click again on "autofollow") is a proof of that. Logically speaking, i believe this bug showcases how this confusion permeates the code.

ocornu, 2025-02-20:
> Thanks for the videos. Only way to communicate these types of dynamic problems.

You're welcome! Notice that i picked the first video hosting website i came across, and it seems they will only remain online for a couple days…

bdbcat, 2025-02-20:
I appreciate your comments.
Doesn't make it right, but this has been the behavior of OpenCPN for at least 5 years.  Just sayin'...
Personally, I find it does have some value "as a hand-tailored version of "lookahead". 
Sometimes an unintentional small chart drag happens, especially on touch screens.  So, on touch screens we do toggle OFF auto-follow on a chart drag.  But not on mouse interfaces.
Have not heard much user discussion on the auto-follow behavior.  I would appreciate a wider discussion on this topic.  Nothing ever set in stone.

ocornu, 2025-02-20:
> Personally, I find it does have some value "as a hand-tailored version of "lookahead".
It does! Now that i know about it, i will very likely use it… :-)
Isn't it a pity that this feature is unavailable on touch screen-devices, just because it's been implemented this way? And that it likely is a feature most users are unaware of (unless they've read the manual)?
I believe both of those issues could be fixed with a better UI design: something explicit, rather than an implicit behavior (only available on certain platforms).…

ocornu, 2025-02-20:
> "better UI design"....Specific suggestions?

Not quite yet, i'm afraid: these things are easier said than done!  :-D
I'll get back to you if i think of something worth mentioning…

bdbcat, 2025-02-20:
Corrected several "corner cases" in source code, I believe this issue ready to close after your test.
Thanks
Dave

ocornu, 2025-02-21:
It seems to behave okay in other modes, but it's [still jumping with lookahead turned ON](https://streamable.com/4zuaue). :-/

ocornu, 2025-02-21:
Thanks for your reactivity and efforts, by the way, Dave. It's much appreciated…  :-)

While i have a hold on you, is there a dedicated place (mailing list, something else) to discuss or ask questions about OpenCPN development?
For example, i'd like to play with the charts a little, and i'm not sure where in the code this is all located. To be more specific, i'd like to remove the CM93 land-filling in order to display an underlying MBTiles layer. I'm assuming the CM93 layer would be transparent if land wasn't filled, but i may be wrong: i've never worked with OpenGL before.
I suppose i'd also have to add my MBTile layer (the one i'd like to appear in-land) between the low-res world map and the CM93 map layers, as current MBTiles are rendered on top of the latter…
Anyway, any hint would be appreciated. Thanks!

bdbcat, 2025-02-21:
ocornu..
The OpenCPN github "Discussion" list is a good place to start.  Post a topic, and you will get some comments.
https://github.com/OpenCPN/OpenCPN/discussions

ocornu, 2025-02-21:
Thanks!

It's a bit of a hack, but i managed on my own in the end…

[image]

rgleason, 2025-02-21:
@ocornu   What dashboard?  (sorry for off topic.)

bdbcat, 2025-02-21:
Don't keep us in suspense.  How did you do that?  :)

ocornu, 2025-02-22:
> [@ocornu](https://github.com/ocornu) I think with bdcat's latest commit to git master both the jumping and range rings issue are solved. Pls test and confirm.

The jumping back and forth with lookahead ON indeed seems to be fixed (0cee838b4e build).

However, i can see two related problems (see [video](https://streamable.com/jtmw7f)) --that may or may not have been there before:
1. when moving a waypoint, opencpn jumps to a previous location of this waypoint (that was far enough to induce "chart scrolling").
2. when zooming/unzooming, one can momentarily drag the boat location away

ocornu, 2025-02-22:
> [@ocornu](https://github.com/ocornu) What dashboard? (sorry for off topic.)

@rgleason: Basic vanilla dashboard, with a couple custom instruments i've been developing… You like it?  :-)

> Don't keep us in suspense. How did you do that? :)

@bdbcat: the satellite view of the land, i assume? Well, pretty much as i've described in the message above:
1. Comment out a line (in s52pclib) to prevent rendering of landmasses
2. Comment out a line (in ChartCanvas) to prevent filling canvas background with "no data" color.
3. Move MBTiles rendering before chart rendering ―therefore underneath (also in ChartCanvas).

It's a tiny patch overall. :-]
Obviously, not suitable for production: lots of corner cases not covered, _etc_. But already enough for me to use (as i'm aware of, and won't bitch about, its limitations). And, to be honest, i mainly see it as a proof of concept for other ideas i have in mind …

rgleason, 2025-02-22:
@rgleason: Basic vanilla dashboard, with a couple custom instruments i've been developing… You like it? :-)

YES!!!  These instruments should certainly be in Dashboard SK and Dashboard. We need these improved graphics desperately, and this style of graphic should be developed for the whole set of instruments. IMHO.

ocornu, 2025-02-22:
@rgleason: Thanks! I wasn't satisfied with the instruments i got to come across (B&G, Simrad, Raymarine…) so i decided to make my own, the way _i_ want them to be for _my_ use case. To be fair, I guess it's easier when you don't have to tailor to a large audience, with (likely) non-homogeneous wishes: no need to restrict oneself to some improbable common denominator. :-p

Integration to the mainstream dashboard might be challenging however as, for example, i've adopted a different philosophy than the one seemingly ruling dashboard_pi: from what i can tell, it aims at doing things very generically, so that one can build a new instrument from a new data source without any code (or close to), with a fine granularity (pretty much one instrument per data source); Whereas i tend to put many source data in one specific instrument, that would probably lose its attractiveness (or break completely) if any one data source was missing… 

Anyway, this is indeed not the place to discuss this further, and i've already blabbed too much.

rgleason, 2025-02-22:
@ocornu  Can you please join the discussion here https://github.com/OpenCPN/OpenCPN/discussions/4341 
I believe your instrument is much like [Tactics_pi  instruments ](https://opencpn-manuals.github.io/main/tactics/index.html)and we would like to consolidate the 4 versions of Dashboard into just one or two, by improving the instruments and adding the missing ones.  I would very much like to get the Tactics instruments incorporated into Dashboard and DashboardSK, perhaps adding your instrument and augmenting it.

bdbcat, 2025-02-22:
Not sure I understand this one:
_when zooming/unzooming, one can momentarily drag the boat location away_

Do you use "Zoom-to-Cursor" mode?

ocornu, 2025-02-23:
> Not sure I understand this one: _when zooming/unzooming, one can momentarily drag the boat location away_
> 
> Do you use "Zoom-to-Cursor" mode?

I do. That"s what's dragging the ship away from its known location while zooming out, as can be seen iirc in the end of the last video…

bdbcat, 2025-02-23:
"That"s what's dragging the ship away from its known location"
Well, of course it does.  The zoom+pan operation resulting from "Zoom-to-cursor" will drag the ship wherever, then the next GNSS update will reposition the chart based on ships lat/lon/COG.  Just as to be expected.  And not especially useful.

I think zoom-to-cursor with auto look-ahead is not a very intuitive mode setup.  I am considering disabling ZTC if look-ahead is active.  Would you object strenuously?

bdbcat, 2025-02-23:
I am considering disabling "Zoom-to-cursor" when auto-follow and look-ahead and valid GNSS position are all active.
I think this set of options is not intuitive, and not particularly useful, since the chart immediately jumps back to the "look-ahead" position on the next GNSS update.  All you have accomplished is a chart zoom, with a distracting jump in the middle.

However, if the look-ahead is not active, then Zoom-to-cursor works as before, and is quite useful sometimes, after getting used to it.
Thoughts?

ocornu, 2025-02-23:
I'm not sure i can trust my imagination for this kind of things: i'd have to test it in the field to be sure. But yeah, it could work…

Side note: with (partial) sea transparency…
[image]

Hakansv, 2025-02-23:
>  am considering disabling "Zoom-to-cursor" when auto-follow and look-ahead and valid GNSS position are all active.

Yes, I'd recommend that logic.

Discussions:
When I'm sailing, I like a zoom function to center around the boat symbol regardless of where the cursor has pulled away on the last heel. For planning and other overview, another zoom center may be suitable.
We have two methods to get zoom center to a specific location. Either a single left click to center and then zoom or enable "zoom to cursor". Both methods are used by individuals and of course, each method is (strongly) preferred by one's own preferences.

Zoom to cursor is though not suitable when auto follow look ahead mode is enabled. Since look ahead mode center the chart back to default view after a chart move operation, like zoom to cursor, Every zoom action will thus cause a "jumping" view port. So better to not use center-to-cursor while in look-ahead mode.

In addition, it is usually not appropriate to create a new route on the same canvas where the look ahead is active. The same bouncing view occurs. In these cases, I usually turn the follow ship function off.

bdbcat, 2025-02-23:
"Yes, I'd recommend that logic."
Done.
Please verify from current github master

bdbcat, 2025-02-23:
With current github master code base, I request closure of this issue.
Thanks
Dave

ocornu, 2025-02-24:
I've tested your last patch, and the new  (look-ahead and zoom-to-cursor ON) workflow does not work for me.
While cruising i used to:
- zoom-in on something ahead i wanted to check-out (say, a cove/inlet i's considering mooring into), and zoom-to-cursor would allow me to do it. There would be no "jumping back" as the boat would have gotten out of the screen by the time i had zoomed-in on my target.
- Then i could explore the area at will…
- and revert back to the boat by clicking on "follow" once i was done.
This is no longer possible, for example.

The new workflow: either unselect "follow" first, or drag the boat out of the map, then zoom-in on your target (if it's still visible after my dragging) is cumbersome and does not feel natural at all.
And having two different behaviors (zoom-to-cursor and zoom-to-boat) for the same action (zooming-in) depending on a tiers modality (whether "follow" is engaged or not) is super confusing: an ergonomic no-go as far as i'm concerned.

I'm sorry i wasn't able to foresee this without testing it, but please, please: revert this patch. 

If you allow me to make my case further ―at the risk of sounding intolerant:
As i see it, zoom-to-cursor is a must-have for any mapping app (not just OpenCPN): ergonomically speaking, it does not make sense to center the zoom on anything else (unless there is no cursor to zoom to, obviously).
Same for look-ahead: in any navigation app tracking a moving vehicle you're in (whether it's a car, a boat or anything else), it does not make sense to _not_ prioritize what's in front; giving as much real-estate to what is behind looks like priorities inversion. Admittedly, OpenCPN is in a bit of a no-man's-land on this front because of its original "free-follow" (or "manual-look-ahead") alternative.
In my opinion, both features should be there and the default on any navigation app (i'd claim: on objective grounds, not just to comply with my own preferences). Thus any such app should prioritize making this combo work and feel seamless, and not cut corners for the sake of playing nice with relic features and old habits.

Hope you will forgive my passion in the above text and not be put off by any over-the-top intensity that might transpire out of it… Thank you.  :-)

bdbcat, 2025-02-24:
Interesting ideas.  Maybe...
Please recast this suggestion as an RFE, for further discussion.
We are well beyond the "map keeps jumping...." title of this issue.

TBH, my initial thought is:  "Oh no, not more user buttons to push.  Many, many users are already overwhelmed by the configurability of OCPN as it is.  They get lost in the options, and finally give up and reload O to get back to something that works for them.  And operate in fear of screwing something up afterwards..."

ocornu, 2025-02-24:
> Interesting ideas. Maybe... Please recast this suggestion as an RFE, for further discussion. 

Alright, will put it in a RFE ―i put it here because that's where you asked.  :-)

> TBH, my initial thought is: "Oh no, not more user buttons to push. Many, many users are already overwhelmed by the configurability of OCPN as it is. They get lost in the options, and finally give up [...]

I get that. What's even more off-putting however is an app you cannot configure to work "as it should" (according to you): that's a definite deal-breaker.
If the default option is chosen carefully, there should be no need to ever confront oneself with these options. And if one needs to, because the app doesn't behave as one wants, one will be glad to find them (and likely not confused, cause one was specifically looking for them).

Remember that, as it is, most people i bet know nothing about the `free-follow` feature, although it could be their favorite if they knew about it. Such a shame…
I'd love to se a survey of what percentage of users know about this feature. Filtering out people who read the whole manual (because you can't expect that from all users), people who've learnt it from another user, _etc_. I mean the percentage of people who've figured it out from using OpenCPN alone (as any correctly designed feature should allow itself to be discovered). I bet it's in the (low) single digits.
I don't mean to take myself as a reference, but i've been playing with computers since i was 7, i've been coding for 40+ years and i wasn't able to figure it out… :-]

bdbcat, 2025-02-24:
"that's a definite deal-breaker."
Strong language, unless there is a better alternative (for you).

bdbcat, 2025-02-24:
Close this issue, please.

ocornu, 2025-02-24:
> "that's a definite deal-breaker." Strong language, unless there is a better alternative (for you).

Actually, no: i was talking about "normal" users. 
As for me, i can always compile my own OpenCPN version without the "smooth dragging" features (i do not need, tbh), but with the look-ahead scroll-to feature i do need. I'll be fine in any case. :-)

ocornu, 2025-02-24:
> "Yes, I'd recommend that logic." Done.

@bdbcat: How is such a decision _not_ subject to a RFE, so that everyone impacted can weigh in, please?
Instead of buried deep into an unrelated bug ticket, and rushed out for the sole reason ―it seems― that you can't bother fixing the bug you introduced?

bdbcat, 2025-02-24:
ocornu...
We generally do not design by "committee", especially UI issues.  What happens if we do that is this: the negotiated solution to be implemented ends up being the "inclusive-or" of everything anyone can think of, or has ever seen anywhere else.  Which leads to a "creep of complexity" in the UI, not beneficial to the average user base.
So, for point issues such as the one discussed here, I make a decision and move ahead.  Lots of fish to fry....